dronekit/dronekit-python #781: Dronekit over 4G

ml1206079, 2018-01-10:
Hey guys.
i have been having a really strange issue i'm using dronekit as a ground-station with a remote server running a drone-kit script which accepts a udpin connection. the drone it self is using mavproxy to transmit the connection over 4G dongle to the ip address of the server running dronekit
sometimes it connects sometimes it doesn't and i have noticed it mostly doesn't connect when i'm indoors.
so i digged in through your code and i found this line 
` msg = self.out_queue.get(True, timeout=0.01)`
line 172 at mavlink.py which i think is causing the problems. because my 4g link indoors has a delay of over 100ms 
do you think i'm right in my assumption ? and if so. why did you guys specifically choose this number as a timeout ?

pietrodn, 2018-03-17:
The line of code you mentioned is not the problem. [`out_queue`is a Queue object](https://github.com/dronekit/dronekit-python/blob/master/dronekit/mavlink.py#L132), which is defined in the [`queue` module](https://docs.python.org/3/library/queue.html#queue.Queue.get).

From the documentation:

> The queue module implements multi-producer, multi-consumer queues. It is especially useful in threaded programming when information must be exchanged safely between multiple threads. The Queue class in this module implements all the required locking semantics. It depends on the availability of thread support in Python; see the threading module.

> `Queue.get(block=True, timeout=None)`
> Remove and return an item from the queue. If optional args block is true and timeout is None (the default), block if necessary until an item is available. If timeout is a positive number, it blocks at most timeout seconds and raises the Empty exception if no item was available within that time. Otherwise (block is false), return an item if one is immediately available, else raise the Empty exception (timeout is ignored in that case).

So, this has nothing to do with network latency.

Can you ping the drone when it does not connect?

borhanreo, 2018-07-06:
I think, you need to use Raspberry Pi and 4G Modem. So you will able to connection between 4G modem,  raspberry pi and socket server. Raspberry pi will connected with APM/PIX via USB cable. I already developed a ground station using 3G modem, raspberry pi 3 B, APM 2.8 and socket server.