dronekit/dronekit-python #782: No waypoint Load started - re-requesting all waypoints

ARPIT355, 2018-01-11:
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 0}
not loading waypoints
not loading waypoints
not loading waypoints
not loading waypoints
not loading waypoints
not loading waypoints
not loading waypoints
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 0}
APM: flight plan received
No waypoint load started
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]

zhydanovajulia, 2018-02-14:
The same issue

ARPIT355, 2018-02-14:
Solved it :-)

Eldradirom, 2018-02-18:
I'm facing the same issue, how did you solve the problem?

ARPIT355, 2018-02-18:
You have to clear mission first and then at last change mode to "AUTO"

Eldradirom, 2018-02-18:
Thank you,
I used this: https://github.com/ArduPilot/MAVProxy/commit/b7765a019224274796c389ad97606a62e463e968
and follow the comment "I did revert this and got rid of the "re-requesting WPs [0, 1, 2, 3, 4]"

It's also working without changing to AUTO mode

swatza, 2018-06-28:
Do we know what the root of this problem actually is? I am running into this problem too.

Phillweston, 2022-05-30:
I am running into this problem too. Haven't solved yet!

MONEEBFAMOUS, 2022-08-05:
I am running SITL with Gazebo9, everything looks fine untill i run mavros in another terminal. After running mavros using the command "roslaunch iq_sim apm.launch", the terminal gets flooded with the wps requaests. Something like this,
STABILIZE> re-requesting WPs [0, 1, 2, 3, 4]
STABILIZE> re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
re-requesting WPs [0, 1, 2, 3, 4]
.
I changed to AUTO mode but the re requesting didn't stop.
I read someone solved this by using "wp list" but i didn't get it, what exactly he did to solve this,