dronekit/dronekit-python #785: Ardupilot APM mega or Pixhawk

ARPIT355, 2018-01-16:
I wanted to know whether dronekit support Pixhwak or Ardupilot APm mega filght controller?

peterbarker, 2018-01-19:
On Tue, 16 Jan 2018, ARPIT355 wrote:

> I wanted to know whether dronekit support Pixhwak or Ardupilot APm mega filght controller?

The mega isn't going to work too well - the code is ancient and doesn't 
support a lot of the things people are wanting to do now-adays.

Works with PixHawk just fine.

ARPIT355, 2018-01-22:
Can i run change attitude and waypoint navigation code of dronekit with Ardupilot APM mega??

peterbarker, 2018-01-22:
On Mon, 22 Jan 2018, ARPIT355 wrote:

> Can i run change attitude and waypoint navigation code of dronekit with Ardupilot APM mega??

Some of it.  A lot of the capabilities (particularly in guided mode) are 
only available on PixHawk or better.

I suggest your time is worth more than the price difference between these 
devices.  *Even* if you already have the mega.