dronekit/dronekit-python #786: Dronekit-python with PX4 and polygonal geofence support

samacumen, 2018-01-19:
Hello,

A cCouple of questions, first of which I might want to raise a CR for.

1) I would like to know if there is "polygonal geofence-support" with dronkit-python? I could not find it being used in the SDK. Is there a plan for its support using dronekit?

2) I am using PX4 flight stack with Pixhawk. How well is it supported? I see most user base running Ardupilot flight stack on Pixhawk.

Thanks!

peterbarker, 2018-01-19:
On Fri, 19 Jan 2018, Shyam Balasubramanian wrote:

>     I would like to know if there is "polygonal geofence-support" with dronkit-python? I could not find it being used in the SDK. Is there a plan for its
>     support using dronekit?

Not really - apart from the fact you can manufacture any mavlink message 
you like.  There's no higher-level-protocol support for it, 'though.

>     I am using PX4 flight stack with Pixhawk. How well is it supported? I see most user base running Ardupilot flight stack on Pixhawk.

Some people definitely have it working - or have had in the past.  There 
are some obvious issues that haven't been addressed - mission waypoint 
numbering springs to mind.  And yes, most people seem to run ArduPilot 
against it.

hamishwillee, 2018-01-29:
DK-Python was designed with ArduPilot in mind and does not work reliably with PX4 for some use cases - specifically missions (things like the fact that DK-Python assumes zero index mission items are the home position). It is a really great API for ArduPilot, but if you wish to use PX4, suggest you instead consider using https://docs.dronecore.io/v/develop/en/