dronekit/dronekit-python #790: Issues installing dronekit-sitl on OS X

kasselvania, 2018-01-22:
Hey Guys,
I've been having an odd issue with my pip install command for dronekit and dronekit-sitl on my OS X computer. For some reason some libraries cannot be altered. I've made sure to be admin and everything, but these are the issues I'm getting back:

```
Peters-MacBook-Pro:~ pk$ pip install dronekit-sitl
Collecting dronekit-sitl
  Using cached dronekit_sitl-3.2.0.tar.gz
Requirement already satisfied: psutil>=3.0 in /Library/Python/2.7/site-packages (from dronekit-sitl)
Requirement already satisfied: dronekit>=2.0.0b6 in /Library/Python/2.7/site-packages (from dronekit-sitl)
Collecting six>=1.10 (from dronekit-sitl)
  Using cached six-1.11.0-py2.py3-none-any.whl
Requirement already satisfied: monotonic==1.2 in /Library/Python/2.7/site-packages (from dronekit>=2.0.0b6->dronekit-sitl)
Requirement already satisfied: pymavlink==2.0.6 in /Library/Python/2.7/site-packages (from dronekit>=2.0.0b6->dronekit-sitl)
Requirement already satisfied: future==0.15.2 in /Library/Python/2.7/site-packages (from dronekit>=2.0.0b6->dronekit-sitl)
Installing collected packages: six, dronekit-sitl
  Found existing installation: six 1.4.1
    DEPRECATION: Uninstalling a distutils installed project (six) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling six-1.4.1:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_set.py", line 778, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_install.py", line 754, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 131, in copy2
    copystat(src, dst)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 103, in copystat
    os.chflags(dst, st.st_flags)
OSError: [Errno 1] Operation not permitted: '/var/folders/lz/gn5m_f594gbfy12vg2qgrdyr0000gn/T/pip-uBbIl8-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/six-1.4.1-py2.7.egg-info'
```

Any tips to correct this?

pietrodn, 2018-01-22:
This is not a `dronekit-sitl` specific problem.
The cause of your problem is [System Integrity Protection](https://support.apple.com/en-us/HT204899), a macOS feature which write-protects `/usr`, `/bin`, `/sbin`, `/System` (not even root can edit or add files in those directories).

So, you cannot install packages in the macOS-provided Python installation in a system-wide fashion.

You have two alternatives:

1. Use a **virtual environment**. You should almost always install packages in a virtual environment in order not to pollute your system-wide Python installation. I recommend the use of [pipenv](https://docs.pipenv.org/), the [officially recommended](https://packaging.python.org/tutorials/managing-dependencies/#managing-dependencies) python packaging tool.
2. **Install a new Python distribution in a writable location.** You can do that easily with [Homebrew](https://brew.sh/), by typing `brew install python2`.
Probably you should do this even if you're following solution n. 1, since you will need to install/upgrade `pip`, `pipenv`, `setuptools`, etc. system-wide.

Alternatively, you may disable SIP, but this is a very bad idea since it would expose you to additional security threats.

kasselvania, 2018-01-25:
I've tried with pipenv to install and run dronekit and dronekit-sitl and I've been running into an issue with this response (btw, I'm new to some of this stuff, so I apologize about these types of questions:

```
Adding dronekit-sitl to Pipfile's [packages]…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv install --skip-lock to bypass this mechanism, then run $ pipenv graph to inspect the situation.
Could not find a version that matches monotonic==1.2,>=1.3
Tried: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 1.0, 1.0, 1.0, 1.1, 1.1, 1.2, 1.2, 1.3, 1.3, 1.4, 1.4
Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv install --skip-lock to bypass this mechanism, then run $ pipenv graph to inspect the situation.
Could not find a version that matches monotonic==1.2,>=1.3
Tried: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 1.0, 1.0, 1.0, 1.1, 1.1, 1.2, 1.2, 1.3, 1.3, 1.4, 1.4
```

Would you know what I would need to do to fix this?

kasselvania, 2018-01-25:
Also, you might have a good idea of how to do this, so, I skipped the lock as the pipenv suggested, and I want to run dronekit py examples I downloaded. Do I need to place the files into the virtual environment's folder? Or can I direct the pipenv to look for that file?

pietrodn, 2018-01-25:
Beware that, by default, pipenv creates a Python 3 virtual environment. DroneKit does not support Python 3 as of now.
Use `pipenv —two` to use Python 2.

pietrodn, 2018-01-25:
`pipenv --two install dronekit_sitl` works like a charm for me on macOS 10.13 with Homebrew's python2:

```
$ pipenv --two install dronekit_sitl
Creating a virtualenv for this project…
Using /usr/local/bin/python2 to create virtualenv…
⠋Running virtualenv with interpreter /usr/local/bin/python2
New python executable in /Users/pietro/.local/share/virtualenvs/pipenv-I8uil235/bin/python2.7
Also creating executable in /Users/pietro/.local/share/virtualenvs/pipenv-I8uil235/bin/python
Installing setuptools, pip, wheel...done.

Virtualenv location: /Users/pietro/.local/share/virtualenvs/pipenv-I8uil235
Installing dronekit_sitl…
Collecting dronekit_sitl
  Downloading dronekit_sitl-3.2.0.tar.gz
Collecting psutil>=3.0 (from dronekit_sitl)
  Downloading psutil-5.4.3.tar.gz (412kB)
Collecting dronekit>=2.0.0b6 (from dronekit_sitl)
  Downloading dronekit-2.9.1-py2-none-any.whl (58kB)
Collecting six>=1.10 (from dronekit_sitl)
  Downloading six-1.11.0-py2.py3-none-any.whl
Collecting monotonic==1.2 (from dronekit>=2.0.0b6->dronekit_sitl)
  Downloading monotonic-1.2-py2.py3-none-any.whl
Collecting pymavlink==2.0.6 (from dronekit>=2.0.0b6->dronekit_sitl)
  Downloading pymavlink-2.0.6.zip (2.6MB)
Collecting future==0.15.2 (from dronekit>=2.0.0b6->dronekit_sitl)
  Downloading future-0.15.2.tar.gz (1.6MB)
Building wheels for collected packages: dronekit-sitl, psutil, pymavlink, future
  Running setup.py bdist_wheel for dronekit-sitl: started
  Running setup.py bdist_wheel for dronekit-sitl: finished with status 'done'
  Stored in directory: /Users/pietro/Library/Caches/pip/wheels/06/e6/f4/4ed88637222525dc9d15941dfcd36e3d2de11887a0ca54c044
  Running setup.py bdist_wheel for psutil: started
  Running setup.py bdist_wheel for psutil: finished with status 'done'
  Stored in directory: /Users/pietro/Library/Caches/pip/wheels/8b/83/e9/f0feab7d059e177e6f3834f42b8a302805fd83dbb16405515a
  Running setup.py bdist_wheel for pymavlink: started
  Running setup.py bdist_wheel for pymavlink: finished with status 'done'
  Stored in directory: /Users/pietro/Library/Caches/pip/wheels/07/35/3a/865a0e9b13e13af29ba72d80c7c604c6efe47645195ec2c3d3
  Running setup.py bdist_wheel for future: started
  Running setup.py bdist_wheel for future: finished with status 'done'
  Stored in directory: /Users/pietro/Library/Caches/pip/wheels/11/c5/d2/ad287de27d0f0d646f119dcffb921f4e63df128f28ab0a1bda
Successfully built dronekit-sitl psutil pymavlink future
Installing collected packages: psutil, monotonic, pymavlink, future, dronekit, six, dronekit-sitl
Successfully installed dronekit-2.9.1 dronekit-sitl-3.2.0 future-0.15.2 monotonic-1.2 psutil-5.4.3 pymavlink-2.0.6 six-1.11.0

Adding dronekit_sitl to Pipfile's [packages]…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
Updated Pipfile.lock (cd59b6)!
```

kasselvania, 2018-01-25:
Hmmm, for some reason monotonic becomes an issue for me with this installation:

```
Using /usr/local/bin/python2 to create virtualenv…
⠋Running virtualenv with interpreter /usr/local/bin/python2
New python executable in /Users/pk/.local/share/virtualenvs/examples-jbiopDKd/bin/python2.7
Also creating executable in /Users/pk/.local/share/virtualenvs/examples-jbiopDKd/bin/python
Installing setuptools, pip, wheel...done.

Virtualenv location: /Users/pk/.local/share/virtualenvs/examples-jbiopDKd
Installing dronekit-sitl…
Collecting dronekit-sitl
Collecting psutil>=3.0 (from dronekit-sitl)
Collecting six>=1.10 (from dronekit-sitl)
  Using cached six-1.11.0-py2.py3-none-any.whl
Collecting dronekit>=2.0.0b6 (from dronekit-sitl)
  Using cached dronekit-2.9.1-py2-none-any.whl
Collecting monotonic==1.2 (from dronekit>=2.0.0b6->dronekit-sitl)
  Using cached monotonic-1.2-py2.py3-none-any.whl
Collecting pymavlink==2.0.6 (from dronekit>=2.0.0b6->dronekit-sitl)
Collecting future==0.15.2 (from dronekit>=2.0.0b6->dronekit-sitl)
Installing collected packages: psutil, six, monotonic, pymavlink, future, dronekit, dronekit-sitl
Successfully installed dronekit-2.9.1 dronekit-sitl-3.2.0 future-0.15.2 monotonic-1.2 psutil-5.4.3 pymavlink-2.0.6 six-1.11.0

Adding dronekit-sitl to Pipfile's [packages]…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv install --skip-lock to bypass this mechanism, then run $ pipenv graph to inspect the situation.
Could not find a version that matches monotonic==1.2,>=1.3
Tried: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 1.0, 1.0, 1.0, 1.1, 1.1, 1.2, 1.2, 1.3, 1.3, 1.4, 1.4
Warning: Your dependencies could not be resolved. You likely have a mismatch in your sub-dependencies.
  You can use $ pipenv install --skip-lock to bypass this mechanism, then run $ pipenv graph to inspect the situation.
Could not find a version that matches monotonic==1.2,>=1.3
Tried: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 1.0, 1.0, 1.0, 1.1, 1.1, 1.2, 1.2, 1.3, 1.3, 1.4, 1.4
```

But another problem for me is if I skip the lock, and the installation goes well and good, how would I install the example packages from github in a way that the virtual environment recognizes the package?

pietrodn, 2018-01-25:
AFAIK those messages are only warnings, your installation is OK.

Which are the "example packages"? Anyway, you can install dependencies in "editable mode" meaning that their installation points to a working directory of a GitHub repo.
See the [pipenv](http://pipenv.readthedocs.io/en/latest/basics/#editable-dependencies-e-g-e) and [pip](https://pip.pypa.io/en/stable/reference/pip_install/#editable-installs) documentation.

If you need a shell "within" the virtual environment, just enter `pipenv shell`.

kasselvania, 2018-01-25:
The example packages I’m talking about are the dronekit examples you can download through the git.
		
		

		Peter Kassel415-722-3417Peter.kassel@me.com

pietrodn, 2018-01-25:
The “example” package is not installed by pipenv/pip into the virtual environment because it is not listed by DroneKit’s setup.py (this is correct, only library files are installed).
To execute the examples, you can clone the repo in the same directory where you did “pipenv install”, and do:

pipenv run python dronekit-python/examples/.../example.py