dronekit/dronekit-python #794: filed building wheel for lxml

moonsummer, 2018-02-02:
I run this command sudo pip install --upgrade droneapi always appear lxml faild and wait long time install.
shoud I use pip install lxml? 
===========================
The directory '/home/farsight/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: droneapi in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: pymavlink>=1.1.50 in /usr/local/lib/python2.7/dist-packages (from droneapi)
Requirement already up-to-date: MAVProxy>=1.4.13 in /usr/local/lib/python2.7/dist-packages (from droneapi)
Requirement already up-to-date: protobuf>=2.5.0 in /usr/local/lib/python2.7/dist-packages (from droneapi)
Requirement already up-to-date: requests==2.5.1 in /usr/local/lib/python2.7/dist-packages (from droneapi)
Requirement already up-to-date: future in /usr/local/lib/python2.7/dist-packages (from pymavlink>=1.1.50->droneapi)
Collecting lxml (from pymavlink>=1.1.50->droneapi)
  Using cached lxml-4.1.1.tar.gz
Collecting pyserial>=3.0 (from MAVProxy>=1.4.13->droneapi)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Requirement already up-to-date: six>=1.9 in /usr/local/lib/python2.7/dist-packages (from protobuf>=2.5.0->droneapi)
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/dist-packages (from protobuf>=2.5.0->droneapi)
Installing collected packages: lxml, pyserial
  Running setup.py install for lxml ... \

error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-1jQTmj/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Z535dk-record/install-record.txt --single-version-externally-managed --compile:
    Building lxml version 4.1.1.
    Building without Cython.
    Using build configuration of libxslt 1.1.28
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/sax.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/__init__.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/cssselect.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/builder.py -> build/lib.linux-armv7l-2.7/lxml
    creating build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-2.7/lxml/includes
    creating build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/html5parser.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/clean.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/diff.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/usedoctest.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/builder.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/defs.py -> build/lib.linux-armv7l-2.7/lxml/html
    creating build/lib.linux-armv7l-2.7/lxml/isoschematron
    copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv7l-2.7/lxml/isoschematron
    copying src/lxml/etree.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/etree_api.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/lxml.etree.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/uri.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv7l-2.7/lxml/includes
[... 10 lines cut ...]
    creating build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/rng
    copying src/lxml/isoschematron/resources/rng/iso-schematron.rng -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/rng
    creating build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl
    copying src/lxml/isoschematron/resources/xsl/XSD2Schtrn.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl
    copying src/lxml/isoschematron/resources/xsl/RNG2Schtrn.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl
    creating build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_dsdl_include.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_svrl_for_xslt1.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_skeleton_for_xslt1.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_message.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_abstract_expand.xsl -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/readme.txt -> build/lib.linux-armv7l-2.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    running build_ext
    building 'lxml.etree' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/src
    creating build/temp.linux-armv7l-2.7/src/lxml
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DCYTHON_CLINE_IN_TRACEBACK=1 -I/usr/include/libxml2 -Isrc -Isrc/lxml/includes -I/usr/include/python2.7 -c src/lxml/etree.c -o build/temp.linux-armv7l-2.7/src/lxml/etree.o -w
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/src/lxml/etree.o -lxslt -lexslt -lxml2 -lrt -lz -lm -o build/lib.linux-armv7l-2.7/lxml/etree.so
    /usr/bin/ld: cannot find -lz
    collect2: error: ld returned 1 exit status
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-1jQTmj/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Z535dk-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-1jQTmj/lxml/

pietrodn, 2018-03-17:
Which hardware platform and operating system are you using?

lxml is a big package and takes a lot of time to build from sources. Try to install it from the binary repositories of your distribution, not from pip.

On Debian, it's as simple as: `sudo apt-get install python-lxml`.

Since you're building on ARM, you may find the [piwheels](https://www.piwheels.hostedpi.com/) project useful, since it has ARM builds of lxml and other packages. That would allow you to install lxml with pip.

If you want or need to build lxml from sources, [this StackOverflow answer](https://stackoverflow.com/questions/3373995/usr-bin-ld-cannot-find-lz) should help you solve the linking problem.

pietrodn, 2018-03-17:
Also, the correct command to install DroneKit is:

    pip install dronekit

(not `droneapi`).