dronekit/dronekit-python #795: collect2: error: ld returned 1 exit status     error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

moonsummer, 2018-02-02:
sudo pip install --upgrade droneapi
The directory '/home/farsight/.cache/pip' or its parent directory is not owned b                                                                                                             y the current user and caching wheels has been disabled. check the permissions a                                                                                                             nd owner of that directory. If executing pip with sudo, you may want sudo's -H f                                                                                                             lag.
Requirement already up-to-date: droneapi in /usr/local/lib/python2.7/dist-packag                                                                                                             es
Collecting pymavlink>=1.1.50 (from droneapi)
Requirement already up-to-date: MAVProxy>=1.4.13 in /usr/local/lib/python2.7/dis                                                                                                             t-packages (from droneapi)
Requirement already up-to-date: protobuf>=2.5.0 in /usr/local/lib/python2.7/dist                                                                                                             -packages (from droneapi)
Requirement already up-to-date: requests==2.5.1 in /usr/local/lib/python2.7/dist                                                                                                             -packages (from droneapi)
Requirement already up-to-date: future in /usr/local/lib/python2.7/dist-packages                                                                                                              (from pymavlink>=1.1.50->droneapi)
Collecting lxml (from pymavlink>=1.1.50->droneapi)
  Using cached lxml-4.1.1.tar.gz
Collecting pyserial>=3.0 (from MAVProxy>=1.4.13->droneapi)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Requirement already up-to-date: six>=1.9 in /usr/local/lib/python2.7/dist-packag                                                                                                             es (from protobuf>=2.5.0->droneapi)
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/dist-pack                                                                                                             ages (from protobuf>=2.5.0->droneapi)
Installing collected packages: lxml, pymavlink, pyserial
  Running setup.py install for lxml ... \^error
    Complete output from command /usr/bin/python -u -c "import setuptools, token                                                                                                             ize;__file__='/tmp/pip-build-h5hcn3/lxml/setup.py';f=getattr(tokenize, 'open', o                                                                                                             pen)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code,                                                                                                              __file__, 'exec'))" install --record /tmp/pip-sBzhul-record/install-record.txt -                                                                                                             -single-version-externally-managed --compile:
    Building lxml version 4.1.1.
    Building without Cython.
    Using build configuration of libxslt 1.1.28
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/sax.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/__init__.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/cssselect.py -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/builder.py -> build/lib.linux-armv7l-2.7/lxml
    creating build/lib.linux-armv7l-2.7/lxml/includes
    copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-2.7/lxml/inc                                                                                                             ludes
    creating build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/html5parser.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/clean.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/diff.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-2.7/lxml/htm                                                                                                             l
    copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-2.7/lxml/ht                                                                                                             ml
    copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/usedoctest.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/builder.py -> build/lib.linux-armv7l-2.7/lxml/html
    copying src/lxml/html/defs.py -> build/lib.linux-armv7l-2.7/lxml/html
    creating build/lib.linux-armv7l-2.7/lxml/isoschematron
    copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv7l-2.7/lxm                                                                                                             l/isoschematron
    copying src/lxml/etree.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/etree_api.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/lxml.etree.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv7l-2.7/lxml
    copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv7l-2.7/lxml/                                                                                                             includes
    copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv7l-2.7/lxml/inclu                                                                                                             des
    copying src/lxml/includes/uri.pxd -> build/lib.linux-armv7l-2.7/lxml/include                                                                                                             s
    copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv7l-2.7/lxml/includ                                                                                                             es
    copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv7l-2.7/lxml/in                                                                                                             cludes
    copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv7l-2.7/lxml/includ                                                                                                             es
    copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv7l-2.7/lxml/inc                                                                                                             ludes
    copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv7l-2.7/lxml/in                                                                                                             cludes
    copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv7l-2.7/lxml/in                                                                                                             cludes
    copying src/lxml/includes/xmlerror.pxd -> build/lib.linux-armv7l-2.7/lxml/in                                                                                                             cludes
    copying src/lxml/includes/config.pxd -> build/lib.linux-armv7l-2.7/lxml/incl                                                                                                             udes
[... cut ...]                                              ad().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" inst                                                                                                             all --record /tmp/pip-sBzhul-record/install-record.txt --single-version-external                                                                                                             ly-managed --compile" failed with error code 1 in /tmp/pip-build-h5hcn3/lxml/

pietrodn, 2018-02-18:
AFAIK the correct package to install is `dronekit`, not `droneapi`:

    sudo pip install --upgrade dronekit

peterbarker, 2018-02-20:
@pietrodn Is correct - you're after dronekit.

The compilation failure is due to a missing library - on a debian-like system this would be solved with `sudo apt-get install -y libz-dev`

closed 2018-02-20 by peterbarker

huangpan2507, 2019-11-21:
> 
> 
> @pietrodn Is correct - you're after dronekit.
> 
> The compilation failure is due to a missing library - on a debian-like system this would be solved with `sudo apt-get install -y libz-dev`

sorry， I am a fresh man . I cann't find a package named  “libzip”， but I use  
>conda search libz* --info  and
>anaconda search -t conda libz* 
and I find the package named  libzip
furthermore  I use 
>conda search libz-*  --info
>anaconda search -t conda libz-* 
these two command found nothing 
so I add the mirror into ~/.condarc 
then >conda install -n my_env_name  libzip-1.5.1

huangpan2507, 2019-11-21:
> 
> 
> @pietrodn Is correct - you're after dronekit.
> 
> The compilation failure is due to a missing library - on a debian-like system this would be solved with `sudo apt-get install -y libz-dev`

sorry， I am a fresh man . I cann't find a package named  “libzip”， but I use  

>conda search libz* --info  

and

>anaconda search -t conda libz* 

and I find the package named  libzip-1.5.1
furthermore  I use 

>conda search libz-*  --info
>anaconda search -t conda libz-* 

these two command found nothing 
so I add the mirror into ~/.condarc 
then 

>conda install -n my_env_name  libzip-1.5.1