dronekit/dronekit-python #797: mavproxy.py command not found

duckhang9113, 2018-02-03:
I'm running ubuntu 17. and install MAVProxy according to the instruction from https://ardupilot.github.io/MAVProxy/html/getting_started/download_and_installation.html. 
I'm not sure what this mean so I skip this step:
If not already set, MAVProxy needs to be on the system path:

echo "export PATH=$PATH:$HOME/.local/bin" >> ~/.bashrc

I try to run mavproxy.py. It returns command not found. Please help me!

pietrodn, 2018-02-18:
Did you reload `.bashrc` in the shell?
You can close and re-open the terminal, or do: `source ~/.bashrc`

hamishwillee, 2018-02-19:
> Also, this is a duplicate of #778.

Yes it is. @duckhang9113 , please continue discussion on #778.

Thanks @pietrodn

closed 2018-02-19 by hamishwillee

duckhang9113, 2018-02-19:
@pietrodn I closed the terminal and re-open it. it's solved. Thank you for your help 🥇 👍 :)