dronekit/dronekit-python #800: Connecting from RPI to Ardupilot 2.8 at 115200 timeout

BillyTziv, 2018-02-14:
Hey I am trying to connect to my Ardupilot 2.8 using a Raspberry Pi 3. I have already installed python 2.7.13. From the directory "/home/pi/dronekit-python/examples/vehicle_state" I am executing the only file there "vehicle_state.py".

It seems that It is not possible to connect.  First of all I have gone through some issues and found many things. By executing the following command
> python vehicle_state.py --connect '/dev/ttyACM0:115200'
it returns the following error
```
Connecting to vehicle on: /dev/ttyACM0:115200
Traceback (most recent call last):
  File "vehicle_state.py", line 40, in <module>
    vehicle = connect(connection_string, wait_ready=False)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2835, in connect
    handler = MAVConnection(ip, baud=baud, source_system=source_system, use_native=use_native)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 128, in __init__
    self.master = mavutil.mavlink_connection(ip, baud=baud, source_system=source_system)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 1235, in mavlink_connection
    return mavudp(device, source_system=source_system, input=input, use_native=use_native)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 875, in __init__
    self.port.bind((a[0], int(a[1])))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.gaierror: [Errno -2] Name or service not known
```

So I removed the baud rate and leave it to default hoping that this is the same. By executing 

> python vehicle_state.py --connect '/dev/ttyACM0'

it returns
```
Connecting to vehicle on: /dev/ttyACM0
>>> Link timeout, no heartbeat in last 5 seconds
```
and I have tried to set the timeout=180 and timeout=300 but nothing worked. I have also tried this fix: [](https://github.com/dronekit/dronekit-python/pull/784/commits/7c0ab3228d844479e9c84df41fd1961170a16803) but it did not work either. So Any ideas how to fix this would be appreciated! Thanks.

BillyTziv, 2018-02-15:
Trying to solve this issue with all my knowledge, I have moved to a new error
`pi@raspberrypi:~/dronekit-python/examples/vehicle_state $ python vehicle_state.py --connect '/dev/ttyACM0'
Connection string found:  /dev/ttyACM0

Connecting to vehicle on: /dev/ttyACM0
>>> Calibrating barometer
>>> ArduCopter V3.2.1 (36b405fb)
>>> Frame: QUAD
>>> barometer calibration complete
>>> GROUND START
Everything went OK! 

Get all vehicle attribute values:
 Autopilot Firmware version: APM:Copter-None.None.NoneUnknownReleaseType
   Major version number: None
   Minor version number: None
   Patch version number: None
   Release type: None
   Release version: None
   Stable release?: False
 Autopilot capabilities
Traceback (most recent call last):
  File "vehicle_state.py", line 57, in <module>
    print("   Supports MISSION_FLOAT message type: %s" % vehicle.capabilities.mission_float)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1678, in capabilities
    return Capabilities(self._capabilities)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 440, in __init__
    self.mission_float                  = (((capabilities >> 0)  & 1) == 1)
TypeError: unsupported operand type(s) for >>: 'NoneType' and 'int'`

Hope this helps someone understand what is going on.

ibrahimhaleemkhan, 2018-11-06:
Did you found out the solution?