dronekit/dronekit-python #801: How to fly plane in dronekit

willbryk720, 2018-02-15:
I'm trying to run the example mission in sitl, by following this link: http://python.dronekit.io/examples/mission_basic.html#example-mission-basic

I run dronekit-sitl plane --home=-35.363261,149.165230,584,353 in terminal 1,
mavproxy.py --master tcp:127.0.0.1:5760 --out 127.0.0:14550 --out 127.0.0.1:14551 in terminal 2,
and python mission_basic.py --connect udp:127.0.0.1:14551 in terminal 3

I view the plane in APM Planner 2.0, and once the simulated vehicle connects it immediately starts spiraling out of control. It says its in manual mode before it switches to AUTO mode, I don't know if the problem might be because of it starting in manual mode?

Once it is spiraling super fast, after a few seconds I get an error: ERROR: Floating point exception - aborting.

Does anyone know what might be going wrong, or better yet does anyone have examples of a working dronekit plane code, because there are no examples using plane anywhere on the dronekit docs or anywhere on the internet, from what I've seen. 

Thanks, and sorry I'm completely new to flying drones.

peterbarker, 2018-02-20:
On Thu, 15 Feb 2018, William Bryk wrote:

> Does anyone know what might be going wrong, or better yet does anyone have examples of a working dronekit plane code, because there are no examples using
> plane anywhere on the dronekit docs or anywhere on the internet, from what I've seen.

I suggest getting used to SITL using ArduPilot's sim_vehicle.py, and then 
add dronekit-python onto that later once you get used to things.

In terms of a Plane auto mission - by avoidance branch flies planes:

https://github.com/peterbarker/dronekit-python/blob/source-system-filtering/examples/avoidance/avoidance-manytests-plane.py#L523

... but the example is horrendously complicated by the parallelism used to 
keep startup times reasonable.

bduva002, 2018-11-13:
Yep, I am having the exact same problem. I will check out the ArduPilot's sim and see how that goes.

hamishwillee, 2018-11-13:
@peterbarker We should either upload a working plane build or remove it and the docs. Currently it is deceptive. I'd rather we got it fixed, but I don't know how make that happen. Thoughts?

peterbarker, 2018-11-13:
On Tue, 13 Nov 2018, Hamish Willee wrote:

> @peterbarker We should either upload a working plane build or remove it and the docs. Currently it is deceptive. I'd rather we got it fixed, but I don't
> know how make that happen. Thoughts?

Ideally we generate SITL binaries on firmware.ardupilot.org and download 
them directly from there.

It's been on my TODO list for a *very* long time - just hasn't been a 
priority.

hamishwillee, 2018-11-14:
> It's been on my TODO list for a *very* long time - just hasn't been a priority.

Fair enough. If it never going to be your priority perhaps we add a note to the docs ?

hamishwillee, 2018-11-15:
@peterbarker I know about the path argument: 
```
dronekit-sitl ./path [args...]  #Start SITL instance at target file location.`
```
But `SITL_BINARY ` is news to me. If you set this to a path, do you just call SITL like `dronekit-sitl [parameters]`?

Anyway, I created https://github.com/dronekit/dronekit-python/pull/874 to try capture this. I didn't come out and say you can't trust plane - just said that some builds are quite old.

closed 2018-11-16 by peterbarker

bduva002, 2018-11-20:
Peter, So I am wondering if I am thinking about this right. It looks like you are creating a list of commands and saving them in the cmds variable. You then upload them via "cmds.upload()". By doing this when you put the plane in AUTO mode it should take off, fly to a way point, then loiter. Is this right?

I am kinda stuck on getting the plane to actually take off or move for that matter. I am using the sim_vehicle.py you suggested above. I attached the code I pulled from your page. 
[Takeing_off_Plane.zip](https://github.com/dronekit/dronekit-python/files/2597598/Takeing_off_Plane.zip)