dronekit/dronekit-python #819: Cannot set to AUTO mode ?

tdard, 2018-04-24:
Good Morning Ladies & Gentlemen,

I have some troubles setting my simulated vehicle in AUTO mode, while I have no problem to change it in another one.
Here is the code:

```
# Import DroneKit-Python
from dronekit import connect, VehicleMode, LocationGlobal, Command
import time
import pymavlink.mavutil as mavutil
#For SITL
import dronekit_sitl
sitl = dronekit_sitl.start_default()
connection_string = sitl.connection_string()

vehicle = connect(connection_string, wait_ready = True)

#To change its mode, the vehicle must be armed
while not vehicle.is_armable:
    print " Waiting for vehicle to initialise..."
    time.sleep(1)

print "\nSet Vehicle.armed=True (currently: %s)" % vehicle.armed 
vehicle.armed = True
while not vehicle.armed:
    print " Waiting for arming..."
    time.sleep(1)
print " Vehicle is armed: %s" % vehicle.armed

print "\nSet Vehicle.mode = AUTO (currently: %s)" % vehicle.mode.name 
vehicle.mode = VehicleMode('AUTO')
while vehicle.mode.name != 'AUTO':
	print " Waiting for changing mode"
	time.sleep(1)
print "Mode: %s" % vehicle.mode.name
```

Here is what I obtain in the Linux Ubuntu shell:
```
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
>>> APM:Copter V3.3 (d6053245)
>>> Frame: QUAD
>>> Calibrating barometer
>>> Initialising APM...
>>> barometer calibration complete
>>> GROUND START
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...

Set Vehicle.armed=True (currently: False)
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
>>> ARMING MOTORS
>>> GROUND START
 Waiting for arming...
>>> Link timeout, no heartbeat in last 5 seconds
>>> ...link restored.
>>> Initialising APM...
 Vehicle is armed: True

Set Vehicle.mode = AUTO (currently: STABILIZE)
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
 Waiting for changing mode
>>> DISARMING MOTORS
 Waiting for changing mode
 Waiting for changing mode
```

When I decide to stop the program (with Ctrl + C), here is the issue that I obtain:
```
^CTraceback (most recent call last):
  File "test_auto_mode.py", line 42, in <module>
    time.sleep(1)
KeyboardInterrupt 
```

Do you have any idea to what happens, and how could I fix it ?

It seems that the simulated copter cannot be set in AUTO mode (despite it is one of its available modes in theory), and because it takes too much time to switch mode (in facts, the issue shows it is almost certain an infinite loop), it becomes disarmed. But I don't know why it cannot be changed.  

I already know that I could use GUIDED mode + refreshing waypoints, but I would like to use the AUTO mode anyway.

Thank you,

Tristan

Ben-Tu, 2018-04-24:
hi,
From my experience the best way is:
set GUIDED mode->arm motor->  takeof to X altitude ->set AUTO mode

tdard, 2018-04-24:
Hey,
Thank you for your quick answer.

I tried what you proposed with adding these lines of code just after arming the vehicle:
```
print "\nSet Vehicle.mode = GUIDED (currently: %s)" % vehicle.mode.name 
vehicle.mode = VehicleMode('GUIDED')
while vehicle.mode.name != 'GUIDED':
	print " Waiting for changing mode"
	time.sleep(1)
print "Mode: %s" % vehicle.mode.name

vehicle.simple_takeoff(40)
while vehicle.location.global_relative_frame.alt < 0.95 * 40:
	print " Waiting for reaching height wanted"
	time.sleep(1)
```
  The vehicle is OK to reach the target altitude, in GUIDED mode. However, when I ask the vehicle to set in AUTO (exactly the same lines of the previous code for changing mode), it does not work (infinite loop...).
Maybe it occurs because no mission is uploaded in the simulated drone, am I correct ? Or the vehicle can be switched to AUTO without having a mission, what do you think ?

Thank you

Ben-Tu, 2018-04-24:
if is no mission in vehicle ,probably is not swich to auto mode,because has nothing to read .
you need add mission by command or upload mission with file
example:
[http://python.dronekit.io/examples/mission_basic.html](url)

Sorry about the english.

tdard, 2018-04-25:
It worked, thank you.

Have a nice day !

amilcarlucas, 2018-04-25:
This can be closed

closed 2018-04-27 by peterbarker

PGSK1234, 2021-07-18:
Hi... I was able to set to "Guided Mode"... but unable to use command on the web host ...any help is much appreciated
[image]