dronekit/dronekit-python #820: Drone won't take off, showing GUIDED mode "unknown"

alfred316, 2018-04-27:
Hello, 

I am trying to run an intel aero RTF drone with some dronekit python code, and I am running into some issues. 

First, if I have wait_read=True in the connection string, then the program times out, so I turn it off. 

The second, is when I have the basic pre-arm checks, it gets stuck on "waiting to initialize drone" and I can't proceed. So, I comment that out to get to the takeoff part. 
However, when I try to change the mode to GUIDED it prints "unknown mode GUIDED", then it proceeds to arm the drone and try to takeoff, but is never able to get off the ground and gives me negative altitude values. Also, this is the mavproxy.py output when the takeoff command is called:
Got MAVLink msg: COMMAND_ACK {command : 22, result : 4}

I'm not sure why the drone will not take off, I have it calibrated and setup as per the steps on the intel aero github page and in QGroundControl I have the minimum throttle under "tuning" set to a fairly high amount, but changing it doesn't seem to have an effect. I would appreciate any insight that could help me find a solution. If you need more information, please let me know. 

Here's the code that I was trying to run (everything up-to the takeoff function call, the rest is irrelevant as it's never reached anyway):
#!/usr/bin/python
from dronekit import connect, VehicleMode, LocationGlobalRelative
import time

vehicle = connect('tcp:192.168.8.1:5760', wait_ready=False)

def arm_and_takeoff(aTargetAltitude):
    """
    Arms vehicle and fly to aTargetAltitude.
    """

    

    print("Arming motors")
    # Copter should arm in GUIDED mode
    vehicle.mode = VehicleMode("GUIDED")
    vehicle.armed = True

    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:
        print(" Waiting for arming...")
        time.sleep(1)

    # Change the parameter value (Copter, Rover)
    #vehicle.parameters['THR_MIN']=100
    
    print("Taking off!")
    vehicle.simple_takeoff(aTargetAltitude)  # Take off to target altitude

    # Wait until the vehicle reaches a safe height before processing the goto
    #  (otherwise the command after Vehicle.simple_takeoff will execute
    #   immediately).
    while True:
        print(" Altitude: ", vehicle.location.global_relative_frame.alt)
        # Break and return from function just below target altitude.
        if vehicle.location.global_relative_frame.alt >= aTargetAltitude * 0.95:
            print("Reached target altitude")
            break
        time.sleep(1)

arm_and_takeoff(10)

ivanalexo, 2018-05-31:
Hi @alfred316 I also have the same issue using the Intel RTF drone and dronekit, Could you solve the problem??

ivanalexo, 2019-04-24:
@xh2415 the problem is that you're using px4 firmware instead of ardupilot, the code from above is for ardupilot because dronekit doesn't have much support for px4. So I recommend you to update the Flight controller firmware of your Intel RTF drone to ardupilot then calibrate it and finally test it. I made it and it works!

yuwei-jpg, 2024-10-11:
When I tried to use guided mode, it kept waiting for the mode to change. I don't understand why.
[image]

hamishwillee, 2024-10-12:
Or use MAVSDK and stick with PX4.

DroneKit was written and tested against ArduPilot. It "worked" with PX4 to some extent https://docs.px4.io/v1.12/en/robotics/dronekit.html but was never optimized for it.

hamishwillee, 2024-10-12:
FYI, the reason is that flight modes were not standardized, so the identity of a particular mode set in PX4 is not the same as the similar named modes in ArduPilot. These are all "custom modes". There is now a MAVLink API for standardizing modes, but ArduPilot does not yet implement it and I have not tested it with PX4.

closed 2024-10-12 by hamishwillee

yuwei-jpg, 2024-10-12:
@hamishwillee Thanks for your advice! I am using the dronekit and dronekit-sitl. And the command is dronekit-sitl copter. I found that my GPS HDOP was up to 121. Will it be the reason that the guided mode won't work?

[image]

yuwei-jpg, 2024-10-13:
@hamishwillee I used the MAVSDK-Python and stuck with PX4 (and PX4 SITL with gazebo). And it succeeded and worked fine.