dronekit/dronekit-python #821: Indoor dronekit

MathiasDePaepe, 2018-04-28:
I'm working on a project to control the Intel Aero RTF drone indoor. The methods of dronekit works with GPS in GUIDED mode, is there a manner to let the drone move and take off in a GPS-denied environment?

peterbarker, 2018-04-29:
On Sat, 28 Apr 2018, Mathias De Paepe wrote:

> I'm working on a project to control the Intel Aero RTF drone indoor. The
> methods of dronekit works with GPS in GUIDED mode, is there a manner to let
> the drone move and take off in a GPS-denied environment?

https://github.com/dronekit/dronekit-python/blob/master/examples/set_attitude_target/set_attitude_target.py#L8

MathiasDePaepe, 2018-04-29:
Thank you! Do i have to execute the python code from my laptop who has connection with the "Aero-xxxxx" WiFi or i it possible to execute the code over ssh connection and move the drone that way without using a SITL?

I tried to execute the drone over ssh connection with the following commands on MobaXTerm:
`drone@drone-CherryTrial:~/Desktop$ python set_attitude_target.py --connect tcp:127.0.0.1:5760`

I heve the following output where initializing keeps going:
```
Connecting to vehicle on: tcp:127.0.0.1:5760
>>> data link #0 lost
>>> data link #0 regained
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
```
And sometimes it stops at 
```
Connecting to vehicle on: tcp:127.0.0.1:5760
>>> data link #0 lost
>>> data link #0 regained
```
where the problem is the wait_ready so i think i have a wrong connection with the drone? I'm using a ssh connection over my home network.

Thanks on advance!

peterbarker, 2018-04-29:
On Sun, 29 Apr 2018, Mathias De Paepe wrote:

> Thank you! Do i have to execute the python code from my laptop who has
> connection with the "Aero-xxxxx" WiFi or i it possible to execute the code
> over ssh connection and move the drone that way without using a SITL?

SITL isn't relevant; it's a simulation environment.  My understanding is 
you're tring to move a real vehicle - so manipulating a virtual one on 
port 5760 isn't going to help :-)

> I tried to execute the drone over ssh connection with the following commands
> on MobaXTerm:
> drone@drone-CherryTrial:~/Desktop$ python set_attitude_target.py --connect
> tcp:127.0.0.1:5760

You're more after

python set_attitude_target.py --connect :14550

(stop the simulation; it may interfere)

supericeandfire, 2020-10-27:
Hi @peterbarker.
I want to ask your indoor code.
Can I set the altitude below 1 meter? Such as 0.5 m.
I don't know my quadrotor one side of motors spin faster than the rest.