dronekit/dronekit-python #823: Unable to view maps in drone-delivery

sarthak268, 2018-05-01:
I am unable to view thee current location of the drone even though, the current location (long and lat) is visible on the screen.

MuhammadFaizanKhan, 2018-10-15:
I even unable to view the map, i have assigned the new token id to map box in `drone_delivery.py `but it saying `{"message":"Account over limit."}`. I guess i have to assigned mapid too but i don't where to get it?

hamishwillee, 2018-10-15:
Sorry, I don't know either. I more pointed this to you as an example of how you might go about it - I don't have time to get it working again.

MuhammadFaizanKhan, 2018-10-17:
I have found the map ID at here https://www.mapbox.com/studio/tilesets/.  It can be
`mapbox.satellite`
`mapbox.terrain-rgb`

Now the map is working fine.

PGSK1234, 2021-07-19:
@MuhammadFaizanKhan I used both the IDs... still see this screen - blank
[image]

Tshabani, 2022-06-08:
> @MuhammadFaizanKhan I used both the IDs... still see this screen - blank... I did not change the token ID though... Is TokenID called ''access_token'? could you please tell me what the 'access_token' should be set to? [image]

Did you get a solution to this?