dronekit/dronekit-python #825: Dronekit connect() takes more than 18sec

GekoCH, 2018-05-03:
Hy

Connecting to a new vehicle via the` connect() ` command takes forever.
Is this normal that it takes more than 15 seconds? Can this be improved?
I tried connecting to an UDP stream from the mavproxy running on the same
machine (udp:127.0.0.1:145550) and also tried to directly connect dronekit
via the Serial/USB port to the PIXHAWK.

Mavproxy on the other hand needs less than 2 seconds. 
(I did try various setting with wait_ready etc but no success)

Andy

closed 2018-05-03 by GekoCH

dgorissen, 2018-05-03:
FYI, I noticed the same thing. Takes +30 seconds on the first connect (to the USB port directly). Subsequent connects are faster.

I noticed you closed soon after opening, found a fix?

GekoCH, 2018-05-03:
sorry yes, the problem was when I try to send a parameter change within the first 15sec to the PIXHAWK when I connected it appears like it would be stuck. But if I wait 15sec and then send the parameter changes it works flawless and dronekit connects within 2 sec to the stream