dronekit/dronekit-python #829: can't upload multiple commands

FSetragno, 2018-05-17:
I am trying to upload a mission from a txt file to my drone. I notice that I can successfully upload a single command, but the script gets stuck to upload() when I try to add multiple commands (loaded from the txt file). Every command from the file can be uploaded individually without any problem.
This is the code I am using:

```
    missionlist = readmission(aFileName, vehicle)

    #Clear existing mission from vehicle
    cmds = vehicle.commands
    cmds.clear()

    for command in missionlist:
        cmds.add(command)

    cmds.upload()
```

kaangoksal, 2018-09-08:
This happens if your waypoint files are impropper (example: lat lon values are wrong)

borhanreo, 2019-06-24:
@kaangoksal You can try this    
    `from dronekit import Command
    def readmission(aFileName, vehicle):
    #print ("\nReading mission from file: %s" % aFileName())
    cmds = vehicle.commands
    missionlist = []
    with open(aFileName) as f:
        for i, line in enumerate(f):
            if i == 0:
                if not line.startswith('QGC WPL 110'):
                    raise Exception('File is not supported WP version')
            else:
                linearray = line.split('\t')
                ln_index = int(linearray[0])
                ln_currentwp = int(linearray[1])
                ln_frame = int(linearray[2])
                ln_command = int(linearray[3])
                ln_param1 = float(linearray[4])
                ln_param2 = float(linearray[5])
                ln_param3 = float(linearray[6])
                ln_param4 = float(linearray[7])
                ln_param5 = float(linearray[8])
                ln_param6 = float(linearray[9])
                ln_param7 = float(linearray[10])
                ln_autocontinue = int(linearray[11].strip())
                cmd = Command(0, 0, 0, ln_frame, ln_command, ln_currentwp, ln_autocontinue, ln_param1, ln_param2,
                              ln_param3, ln_param4, ln_param5, ln_param6, ln_param7)
                missionlist.append(cmd)
    return missionlist

    def download_mission(vehicle):
    print(" Download mission from vehicle")
    missionlist=[]
    cmds = vehicle.commands
    cmds.download()
    cmds.wait_ready()
    for cmd in cmds:
        missionlist.append(cmd)
    return missionlist

    def save_mission(aFileName,vehicle):
    missionlist = download_mission(vehicle)
    output='QGC WPL 110\n'
    home = vehicle.home_location
    output+="%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\n" % (0,1,0,16,0,0,0,0,home.lat,home.lon,home.alt,1)
    for cmd in missionlist:
        commandline="%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\t%s\n" % (cmd.seq,cmd.current,cmd.frame,cmd.command,cmd.param1,cmd.param2,cmd.param3,cmd.param4,cmd.x,cmd.y,cmd.z,cmd.autocontinue)
        output+=commandline
    with open(aFileName, 'w') as file_:
        print(" Write mission to file")
        file_.write(output)
    class DroneMissionClass:
        @staticmethod
        def upload_mission( aFileName,vehicle):
        missionlist = readmission(aFileName,vehicle)
        print (' Clear mission')
        cmds = vehicle.commands
        cmds.clear()
        # Add new mission to vehicle
        for command in missionlist:
            cmds.add(command)
        print (' Upload mission')
        vehicle.commands.upload()
        return True
    @staticmethod
    def set_drone_home_location(vehicle,alt):
        my_location_alt = vehicle.location.global_frame
        my_location_alt.alt = alt
        vehicle.home_location = my_location_alt
        print(" New Home Location (from attribute - altitude should be 222): %s" % vehicle.home_location)`

ahmmyr, 2020-08-20:
if your commands are properly setup and formatted, you can try to resolve this by doing a vehicle.wait_ready(True, raise_exception=False) before doing your cmds.upload().

i had the same issue and that's what fixed it for me.

ForrestFire0, 2022-04-12:
I just ran into the symptoms of my error - however mine was a different cause. I would like to share in case someone else tries to do the same thing. If you try to upload missions in a callback set by vehicle.add_message_listener, the main thread (which I assume usually tends to the uploading) is blocked until the function returns. You need to have the callback set a flag that another thread can tend to.

TLDR - don't set missions in a callback - it blocks the dronekit thread.