dronekit/dronekit-python #831: About navigation accuracy

vincentwen1995, 2018-05-22:
Hi. I noticed that during simulation and real test on the drone, when using the navigation function goto() that implements the MAVLink commands MAV_CMD_NAV_WAYPOINT, the drone cannot reach the exact location of the waypoint and hangs around 10 cm away from the waypoint without further approaching it. As our application needs some accurate actuation, I was wondering where the problem lies in this case and what direction can be looked into to solve it? On the real drone, an EMILD RTK GPS module is used and RTK fix is achieved, so accuracy should not be the problem.

hamishwillee, 2018-05-22:
This is a question for the autopilot - we just send it a message. I guess there is a small chance that there is a rounding error somewhere on the types used. You could easily test this by trying the same message from a GCS and seeing if you get different behaviour.

vincentwen1995, 2018-05-22:
Thank you much.

closed 2018-05-22 by vincentwen1995