dronekit/dronekit-python #832: Using Dronekit with PX4

JorgeG30, 2018-05-26:
Hello, I was wondering how much of a difference is it when using Dronekit with PX4 installed. Currently I have APM:Copter 3.5.3 installed but PX4 supports more mavlink commands like SET_MESSAGE_INTERVAL so I was wondering what were the potential drawbacks when using PX4 instead of Ardupilot. From what I saw online it seems that there is support for basic mission handling and vehicle monitoring. I would just like to know of any disadvantages to using PX4 instead of ardupilot

peterbarker, 2018-05-27:
On Fri, 25 May 2018, JorgeG30 wrote:

> Hello, I was wondering how much of a difference is it when using Dronekit
> with PX4 installed. Currently I have APM:Copter 3.5.3 installed but PX4

My understanding is that dronekit-python will work against either stack at 
this point.  It is more tested against ArduPilot.

hamishwillee, 2018-05-28:
ArduCopter does not support interval by time (as far as I know), but it does by distance (within missions). 

DK-Python is great for ArduPilot, but it isn't formally tested at all with PX4 - and could well be broken. The development of DK-Python also had ArduPilot in mind, and there are some aspects of the design that reflect this - particularly in mission handling. Upshot, I wouldn't use it for PX4 unless you have no choice.

For PX4 you can consider DroneCore if you know C++ (Swift, Android Java, and Python bindings to follow).

JorgeG30, 2018-05-28:
Thank you both for the information. I appreciate it.

pietrodn, 2018-10-20:
I am currently using DroneKit-Python in production with PX4. There are some problems with parameter handling (DroneKit implements the APM variant of the [parameter protocol](https://mavlink.io/en/protocol/parameter.html) which is not compatible with the PX4 implementation) and some PX4 features are unsupported, but you can cope by directly sending MAVLink messages.

affourtit2, 2018-11-29:
@pietrodn Hello, I am trying to upload waypoint missions to a PX4 using Dronekit but I can't avoid errors with being unable to set the "Planned Home" location. Have you coped with this specific problem by directly sending MAVlink messages?

Specifically, a workaround for this error: WP index out of bounds

If you have, can you point me towards a resource to help me?

Thank you!

pietrodn, 2018-11-29:
I have no experience with missions in PX4, but maybe if you share the code, the expected results, the precise error message you are getting, and a traceback, then someone may be able to help you. :-)