dronekit/dronekit-python #836: Reading 'unsupported' vehicle state attributes?

gpetty, 2018-06-08:
There are numerous state attributes of the Pixhawk2.1/Cube (APM:Copter-3.5.5) that Mission Planner displays in its status page that aren't included in the list of [supported attributes using dronekit](http://python.dronekit.io/guide/vehicle_state_and_parameters.html).  As just one example, I would like to be able to read the 'press_abs' variable  (pressure reported by the internal barometer).  Is there any way to do this with the official release of dronekit?

hamishwillee, 2018-06-09:
Yes. you can directly listen for the message of interest using http://python.dronekit.io/guide/mavlink_messages.html

You can also subclass vehicle to create your own API with the attribute of interest - http://python.dronekit.io/examples/create_attribute.htm

gpetty, 2018-06-09:
Thanks.  This was exactly the lead I needed.  Now a followup question:  I'm looking at the raw MAV messages (wildcard listener; see attached screen cap), and while I understand what many of them are, some are more obscure.  Is there a comprehensive reference somewhere for the interpretation of each variable?  

For example, in the SCALED_PRESSURE message, what is 'press_diff' a difference _from_?

Also, it appears that absolute GPS time is not among the values reported, only system time, which could be wrong if the internal clock hasn't been recently set  -- is this correct?

[image]

hamishwillee, 2018-06-09:
The definitions are here: https://mavlink.io/en/messages/common.html#SCALED_PRESSURE

If there isn't enough information then best place to ask questions is the MAVLink slack channel

gpetty, 2018-06-10:
Perfect, thank you.

It's still a bit mystifying that there's apparently no way to capture the GPS-reported time; my cube only reports microseconds since boot.

closed 2018-06-11 by hamishwillee