dronekit/dronekit-python #837: Problems with source-system-filtering branch

patrick-10, 2018-06-27:
Hello everyone,

I'm new to git, so this might not be the right place for my problem...
I'm currently trying to implement the source-system-filtering branch of dronekit python by @peterbarker for use with my custom multivehicle groundstation to transition to a single-antenna setup (instead of individual radio pairs for each drone-ground-link). 
I used the avoidance example for guidance on how to install this dronekit version, but the line 
`git remote add peterbarker https://github.com/peterbarker/dronekit-python`
does not work for me (`fatal: Not a git repository (or any of the parent directories): .git
`), so I tried to install it manually from the downloaded zip-file.
Also, I'm not using SITL but a serial connection via a pair of SiK radios to one ArduPilot board, so I had to modify avoidance.py from the example a bit. The radios are RFD868x with the asynchronous multipoint firmware. Manual transmission of bytes between the radios does work however.

My code:

   #
    import sys, time, math, struct
    from operator import attrgetter
    import json # to read config files
    import imp # to use external mission modules in python code

    # pymavlink interface for MAVLink communication
    import pymavlink.mavutil as mavutil

    # Additional outsourced code
    from gui_05 import * 
    from util import *
    from swarm_log import *

    # Map module
    import mavproxy_map_adapted as MPM
    from mavproxy_map_adapted import mp_slipmap

    # Port to Dronekit
    import dronekit 
    import mavlink_hub

    # Multithreading
    import threading 
    import Queue

    hub_thread = None
    target_systems = [1]

    hub_connection_strings = []

    # dronekit-python (us!) port:
    connection_string = "/dev/ttyUSB0" #"udpout:localhost:2345"
    #hub_connection_strings.append("udpin:localhost:2345")

    hub_should_quit = False
    def mavlink_hub_target():
        print("Hub thread starting")
        hub = mavlink_hub.MAVLinkHub(hub_connection_strings)
        hub.init()
        while not hub_should_quit:
            hub.loop()
        print("Hub quitting")
        hub.connection_maintenance_target_should_live = False

    print("Starting mavlink_hub thread")
    hub_thread = threading.Thread(target=mavlink_hub_target)
    hub_thread.start()

    # Create a MAVLink connection:
    print("Attempting connect to (%s)" % (connection_string,))
    mav = dronekit.mavlink.MAVConnection(connection_string)
    print("Connected")

    #time.sleep(10)

    vehicles = [None] * len(target_systems)
    # Connect to the Vehicle
    def vehicle_connect_target(mav, id, offset):
        vehicle = dronekit.connect(mav, wait_ready=True, target_system=id)
        vehicles[offset] = vehicle

    offset = 0
    connect_threads = []
    for id in target_systems:
        print('Connecting to vehicle with system ID (%d)' % id)
        connect_threads.append(threading.Thread(target=vehicle_connect_target,
                                                        args=(mav,id,offset)))
        connect_threads[-1].start()
        offset += 1
        time.sleep(2) # mav is not thread safe?!

    for thread in connect_threads:
        print("Waiting for connect thread...")
        thread.join()

    def dump_vehicle_state(vehicle):
        print(" Autopilot Firmware version: %s" % vehicle.version)
        print("   Major version number: %s" % vehicle.version.major)
        print("   Minor version number: %s" % vehicle.version.minor)
        print("   Patch version number: %s" % vehicle.version.patch)
        print("   Release type: %s" % vehicle.version.release_type())
        print("   Release version: %s" % vehicle.version.release_version())
        print("   Stable release?: %s" % vehicle.version.is_stable())

    for vehicle in vehicles:
        dump_vehicle_state(vehicle)

    if hub_thread is not None:
        hub_should_quit = True
    hub_thread.join()

However, after connecting to the vehicle I only receive mav recv errors and a link timeout:
`mav recv error: invalid MAVLink CRC in msgID 243 0xf818 should be 0x4df8`
`mav recv error: invalid MAVLink prefix '120'`
`mav recv error: invalid MAVLink CRC in msgID 0 0xfb60 should be 0x5203`
`mav recv error: invalid MAVLink prefix '96'`
`mav recv error: invalid MAVLink CRC in msgID 192 0x7e00 should be 0xa5c7`
`mav recv error: invalid MAVLink prefix '3'`
`mav recv error: invalid MAVLink CRC in msgID 0 0x18fe should be 0x726b`
`mav recv error: invalid MAVLink prefix '255'`
`mav recv error: unknown MAVLink message ID 15`
`mav recv error: invalid MAVLink prefix '128'`

Does anyone know what could be the cause of these errors? Any advice is very much appreciated!
Patrick

peterbarker, 2018-07-04:
On Wed, 27 Jun 2018, patrick-10 wrote:

> However, after connecting to the vehicle I only receive mav recv errors and a link timeout:
> mav recv error: invalid MAVLink CRC in msgID 243 0xf818 should be 0x4df8
> mav recv error: invalid MAVLink [...]

That looks like a very dirty link - or possibly somebody else using the 
serial port at the same time as pymavlink.

btsatoukzn, 2022-03-02:
hi,

I got a similar error. 
I think baud rate is different.
check fc parameters.

have a good day.