dronekit/dronekit-python #841: set battery values in sitl

mehboobyounus, 2018-08-06:
how to change battery values in sitl copter simulation in one connection through dronekit python api..?

NassimSadallah, 2018-09-03:
hi, i have the same problem, after 2 minutes of flying i get a warning message on the console about the battery voltage. So if anyone know how to resolve this problem we are waiting for...
changing the battery values i think there is no manner, i've searched for this solution and i didn't found anything, but is there a way to modify the battery type (in the sitl there is a 3S with low mAH i think) if just we can change for exemple 4S with 5000mAh or something like.....

ShockvaWe, 2019-05-09:
Hello, I am working on the same thing too.
I found default.parm files under sitl but there is nothing about battery except BATT_MONITOR and I don't think that its related to the battery type or anything, so if someone can help it would be really great.

BkMak16, 2020-03-22:
Hi

You can change the battery values on the ArduPilot SITL after starting the SITL.
Use the param command.
In the SITL terminal start with "param show SIM_BATTERY*" no quotations, astrix at the end, and it should show your current battery voltage.

Change the value with param set SIM_BATTERY {VALUE}