dronekit/dronekit-python #842: Problem With Connecting MAVProxy to Mission Planner (Might be problem with MAVProxy Code)

jasonscupham, 2018-08-06:
Hi Everyone, I have been following a tutorial on YouTube on how to connect mavproxy to mission planner but I'm having problems. I'm typing in:
 C:\Users\Jason Scupham>mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14551 --out udp:10.55.222.120:14550 

I press enter and get this:

Connect tcp:127.0.0.1:5760 source_system=255
Running script (C:\Users\Jason Scupham\AppData\Local\MAVProxy\mavinit.scr)
Running script C:\Users\Jason Scupham\AppData\Local\MAVProxy\mavinit.scr
-> set moddebug 2
-> module load help
Traceback (most recent call last):
  File "C:\Python27\Scripts\mavproxy.py", line 294, in load_module
    m = import_package(modpath)
  File "C:\Python27\Scripts\mavproxy.py", line 435, in import_package
    mod = __import__(name)
  File "C:\Python27\lib\site-packages\MAVProxy\modules\mavproxy_help.py", line 11, in <module>
    import wxversion
ImportError: No module named wxversion

Traceback (most recent call last):
  File "C:\Python27\Scripts\mavproxy.py", line 294, in load_module
    m = import_package(modpath)
  File "C:\Python27\Scripts\mavproxy.py", line 435, in import_package
    mod = __import__(name)
ImportError: No module named help

Failed to load module: No module named help. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Unknown command 'graph timespan 30'
Log Directory:
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
online system 1
STABILIZE> Mode STABILIZE
fence breach
APM: APM:Copter V3.3 (d6053245)
APM: Frame: QUAD
Flight battery 100 percent
Received 526 parameters
Saved 526 parameters to mav.parm

But it just stops here for some reason, tried uninstalling and reinstalling mavproxy, dronekit and dronekit-sitl but it still comes up with this, I'm Running this on windows 10. Any help would be greatly appreciated.
Cheers Jason

peterbarker, 2018-08-14:
On Mon, 6 Aug 2018, jasonscupham wrote:

> ImportError: No module named wxversion

Looks like you need python-wx or something here.

> Failed to load module: No module named help. Use 'set moddebug 3' in the MAVProxy console to enable traceback
> Unknown command 'graph timespan 30'
> Log Directory:
> Telemetry log: mav.tlog
> MAV> Waiting [...]

Have you tried pressing enter a few times?  You should get a "STABILIZE>" 
prompt....

jasonscupham, 2018-08-15:
Hi Peter, sorry for the late reply, I restarted my computer and it seems to have solve the issue. But however I did a prompt saying that yes.

However The import error still occurs. I have tried to install the latest version of wxversion but it still comes up as an error for some reason?

peterbarker, 2018-08-16:
@jasonscupham OK, so if you have the prompt then MAVProxy is generally working.  `help`, `status` etc should all work.

Looks like you're using Windows - sorry, I can't help much there!

You might try asking on ArduPilot's MAVProxy gitter channel - https://gitter.im/ArduPilot/MAVProxy

jasonscupham, 2018-08-16:
Right, Cheers Peter, no worries i'll have a look at the link now

closed 2018-08-16 by jasonscupham