dronekit/dronekit-python #843: DroneKit DLL for Unity project

EstebanGameDevelopment, 2018-08-13:
Hello,

I'm working in a project which has the goal to be able to control a drone from a [Unity](https://unity3d.com/) application.  The idea is to use a [360 camera](https://www.amazon.com/Insta360-Camera-Air-Micro-USB/dp/B01MT9BRNL/ref=sr_1_7?s=electronics&ie=UTF8&qid=1534175551&sr=1-7&keywords=insta360) placed on the top of the drone and through remote streaming and a [VR Cardboard headset](https://vr.google.com/daydream/smartphonevr/) being able to control it remotely. The Unity part is completed. I also got running the Python script that allow me to change the velocity of the drone and I've been able to run it in the MAVProxy emulator successfully. Also, I've been working with [IronPython](https://github.com/IronLanguages/ironpython2/releases/tag/ipy-2.7.8) and I've been able to run Python scripts from the Unity application. Right now the only missing piece of this project is that I need a way to get DroneKit as a DLL so I can import it the same way I'm doing right now with IronPython. My knowledge of making DLLs is limited so I would like to ask for support in that matter. Does anyone can help me how to get DroneKit as a DLL so I can use it?

Thanks

peterbarker, 2018-08-16:
On Mon, 13 Aug 2018, Esteban Gallardo wrote:

> I'm working in a project which has the goal to be able to control a drone from a Unity application. The idea is to use a 360 camera placed on the top of
> the drone and through remote streaming and a [...]

dronekit-python is just Python scripts/libraries...

EstebanGameDevelopment, 2018-08-16:
Thanks for replying Peter.

I've tried to load dronekit-python without luck like this:

`	var engine = Python.CreateEngine();

	ScriptSource scope = engine.CreateScope();

	ICollection<string> sp = engine.GetSearchPaths();

	// Path to the DroneKit scripts
	string path = Path.Combine(Path.GetDirectoryName(UnityEngine.Application.dataPath), "Assets/DroneKit");

	sp.Add(path);
	engine.SetSearchPaths(sp);
	foreach (string item in sp)
	{
		Debug.LogError("+++++item=" + item);
	}
	engine.Runtime.LoadAssembly(System.Reflection.Assembly.GetAssembly(typeof(GameObject)));`

If my post is bothering the community feel free to delete it. I'm here to contribute not to bother you. Usually when there is nobody willing to help me I got no response at all and that's ok for me. I'm used to do the things without help. The main problem here is that I've wasted 3 days in this and I really don't want to reinvent the wheel when someone out there knows the solution that would allow me to solve this issue in a matter of minutes.

peterbarker, 2018-08-16:
On Thu, 16 Aug 2018, Esteban Gallardo wrote:

> I've tried to load dronekit-python without luck like this:

You might want to paste some errors in.  I'm not capable of helping you 
with this, sorry...

> If my post is bothering the community feel free to delete it. I'm here
> to contribute not to bother you. Usually when there is nobody willing to
> help me I got no response at all and that's ok for [...]

Understood.... what you're doing is a little unusual, but maybe something 
will happen.

EstebanGameDevelopment, 2018-08-16:
I need to run this code. Basically it's about sending velocity parameters to the Drone. I don't need anything else but that. I have been able to run successfully the Python script with the interpreter command console and the MAVProxy. When I try to run this same script within Unity thanks to IronPython DLL I got the error message that he is unable to find "dronekit". I've tried to put the whole Dronekit library there but I got stuck in a module called "path". I have not been able to progress in this option, now I'm trying to do it with Dronekit-Android but I'm facing also terrible problems:

`import time
from dronekit import connect, VehicleMode, LocationGlobalRelative, Command, LocationGlobal
from pymavlink import mavutil

#-- Connect to the vehicle
print('Connecting...')
vehicleactivated = True
vehicle = connect('udp:127.0.0.1:14551')

#-- Setup the commanded flying speed
gnd_speed = $TOTALSPEED # [m/s]

#-- Define arm and takeoff
def arm_and_takeoff(altitude):

   while not vehicle.is_armable:
      print("waiting to be armable")
      time.sleep(1)

   print("Arming motors")
   vehicle.mode = VehicleMode("GUIDED")
   vehicle.armed = True

   while not vehicle.armed: time.sleep(1)

   print("Taking Off")
   vehicle.simple_takeoff(altitude)

   while True:
      v_alt = vehicle.location.global_relative_frame.alt
      print(">> Altitude = %.1f m"%v_alt)
      if v_alt >= altitude - 1.0:
          print("Target altitude reached")
          break
      time.sleep(1)
      
 #-- Define the function for sending mavlink velocity command in body frame
def set_velocity_body(vehicle, vx, vy, vz):
    msg = vehicle.message_factory.set_position_target_local_ned_encode(0,0,0,mavutil.mavlink.MAV_FRAME_BODY_NED, 0b0000111111000111, 0, 0, 0, vx, vy, vz, 0, 0, 0, 0, 0)
    vehicle.send_mavlink(msg)
    vehicle.flush()
    
#-- Key event function
def set_new_direction(nvx, nvy, nvz):
	print("Select new speed")
	set_velocity_body(vehicle, gnd_speed * nvx, gnd_speed * nvy, gnd_speed * nvz)
    
#---- MAIN FUNCTION
#- Takeoff
arm_and_takeoff(10)

#- New direction
returnhome = $RETURNHOME;
if returnhome == 1:
	print("Returning to Launch")
	vehicle.mode = VehicleMode("RTL")
else:
	k = 0
	while k < $TOTALTIME:
		set_new_direction($PARAMVX,$PARAMVY,$PARAMVZ)
		# sleep so we can see the change in map
		time.sleep(1)
		k = k + 1
	
# Close vehicle object before exiting script
print("Close vehicle object")
vehicle.close()

vehicleactivated = False`

EstebanGameDevelopment, 2018-11-13:
> Hey @EstebanGameDevelopment perhaps some context into your environment would go a long way, why are you trying to control a drone from Unity?

Yes, sure. Here you have the project in a nice video description. Basically it's exactly the same I've explained in the previous posts of this same thread. I was unable to run it on a real drone, but as soon as I earn enough money to invest in a drone or find someone interested to help me. I suppose it will take me a year to get my hands on the drone I need to complete the full project, in the meantime anybody with resources can complete the project and get all the benefit of my work for themselves:

https://assetstore.unity.com/packages/essentials/tutorial-projects/360-remote-drone-controller-127290