dronekit/dronekit-python #847: Detecting types of waypoints in Mission Planner

jasonscupham, 2018-08-25:
Hi Everyone,
I have had a look at some of the functions in the mission planner, a group especially caught my eye which was the VTOL commands, I was wondering how that could be programmed into Dronekit script?
Also when a VTOL takeoff command is called. Is there a way to detect that is happening and react to this via a if statement or something else?
cheers Jason

hamishwillee, 2018-08-26:
YOu can send any MAVLink message you like using `message_factory` - there is an example here: http://python.dronekit.io/guide/copter/guided_mode.html?highlight=message_factory

I don't know whether the takeoff command for VTOL returns an ACK or not. However you could monitor for the altitude and then send a new command when you're happy that you've reached the desired altitude. The docs show you have to get telemetry.

hamishwillee, 2018-08-26:
PS. Best place to ask questions is the gitter channel.

closed 2018-08-26 by hamishwillee