dronekit/dronekit-python #848: Altitude drop when mode is changed from GUIDED to LOITER

anilkunchalaece, 2018-08-28:
When vehicle mode is changed from GUIDED to LOITER , the vehicle altitude is droping to 0 and vehicle crashes.

the code i'm trying is 
```
    while vehicle.mode.name == "GUIDED" : #stop action if we are no longer in guided mode
        remainingDistance = getDistanceInMeters([vehicle.location.global_relative_frame.lat,vehicle.location.global_relative_frame.lon],current)
        print("distance to target :: %s "%remainingDistance)
        if remainingDistance <= targetDistance * 0.01 : #just below target
            print "reached target"
            break
        sleep(2)
    #change the mode to loiter and wait for 10 seconds
    vehicle.mode = VehicleMode("LOITER")
    while not vehicle.mode.name == "LOITER":
        print "Waiting for vehicle to change to LOITER Mode"
        sleep(1)
    print("current vehicle mode is : %s"% vehicle.mode.name)
    ##Wait for 10 sec
    sleep(5)
```

do i have to set altitude before changing mode ?

anilkunchalaece, 2018-08-29:
Looks like it can be avoided by setting the throttle to 1500 (after takeoff ?)
```
armAndTakeOff()
#set the channel overrides which avoids crashing when mode changed to loiter
print "setting throttle channel to 1500 via channel overrides"
vehicle.channels.overrides['3'] = 1500
```
I dont know is this proper way or not, since[ dronekit doc](http://python.dronekit.io/examples/channel_overrides.html) highly dis-commended using rc overrides

hamishwillee, 2018-08-29:
I'd be looking at why the loiter is causing a crash. Possibly the vehicle is transitioning from guided to a manual mode before you enable loiter - and your throttle is physically low on the RC control. There will be a reason this is happening (ie a parameter or similar).  This is an autopilot question - but you could do some debug by looking at the telemetry after reaching guided. 

Another possibility is that the set loiter altitude is too low.

anilkunchalaece, 2018-08-30:
I'm testing this code in simulator (dronekit-sitl)
this is the mavproxy output
```
GUIDED> Mode GUIDED
Arming checks disabled
Got MAVLink msg: COMMAND_ACK {command : 22, result : 0}
GPS lock at 0 meters
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 0}
Got MAVLink msg: COMMAND_ACK {command : 178, result : 0}
Flight battery 90 percent
height 10
Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
LOITER> Mode LOITER
height 0
Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}
GUIDED> Mode GUIDED
```
Loiture altitude is too low ?
how can we set that ? is there any command in dronekit ?

lachlanbrewster, 2020-07-26:
Any change on this? I get the same problem, I imagine that it happens because of no rc input, and using `vehicle.channels.overrides['3'] = 1500`
Doesn't help, does this mean that we just can't use loiter mode when running simulations?

benjamin2044, 2021-07-08:
I literally crashed my drone when I changed from the GUIDED mode to ALTHOLD mode.  My initial guess was ~ may be I have some problem with the Pixhawk. However, when I simulated a Copter with SITL and found the same behavior when I observed in the Mission Planner. Only mode that worked was RTL and LAND.
The lesson I learned, never changed to other modes while you are in GUIDED mode except to RTL or LAND.

benjamin2044, 2021-11-09:
> Any change on this? I get the same problem, I imagine that it happens because of no rc input, and using `vehicle.channels.overrides['3'] = 1500` Doesn't help, does this mean that we just can't use loiter mode when running simulations?
> 
> EDIT: using the rc override every iteration of my loop (~0.1 seconds) means that I can use loiter.

I think we have to make a separate thread in the python dronekit script where it's continuously running the code
vehicle.channels.overrides['3'] = 1500
and use rest of the other code to something else.

wangcongai, 2024-05-15:
> I think we have to make a separate thread in the python dronekit script where it's continuously running the code vehicle.channels.overrides['3'] = 1500 and use rest of the other code to something [...]

I impliment this thread idea, and successfully maintain the height of the copter after changing GUIDED MODE to STABILIZE MODE. **The mechanisim of it is, every time we do assignment "vehicle.channels.overrides['3'] = 1500", dronekit calls mavlink.rc_channels_override_send function to set RC channel value**. We have to KEEP DOING the assignment "vehicle.channels.overrides['3'] = 1500", to get the motor spinning steadily.
Hers is the implimentation:
`channel_lock = threading.Lock()
stop_rc_thread = False
rc_override_values = [1500, 1500, 1363, 1500] 
def override_loop():
    global stop_rc_thread, rc_override_values
    while not stop_rc_thread:
        with channel_lock:
            vehicle.channels.overrides['1'] = rc_override_values[0]
            vehicle.channels.overrides['2'] = rc_override_values[1]
            vehicle.channels.overrides['3'] = rc_override_values[2]
            vehicle.channels.overrides['4'] = rc_override_values[3]
        time.sleep(0.1)
override_thread = threading.Thread(target=override_loop)
override_thread.start()
set_mode('STABILIZE')`