dronekit/dronekit-python #849: How can I arm and takeoff the drone without remote controll using Pixhawk and raspberry by dronekit python

Bashir1998, 2018-08-30:
We have bought a Pixhawk 2.1 flight controller , and we are working on a university project ; the project is that we have to solve a puzzle by a drone ( which contains a raspberry pi 3 ) and send the right way coordinates to a small car . 
Our problem is that we shouldn’t use any remote controller on the drone . So the drone should fly to a specific altitude and then should start looking for the puzzle and solve it .  

We have tried to use dronekit python . We could connect to the drone by the code but we couldn’t make it arm and take off .

Is Pixhawk 2.1 support arming and take off with out using any ground station or remote control? . If it does we hope that you send us a code and the method . Because we have search a lot and the project dead line is very near .

Thank you for your time ...

hamishwillee, 2018-08-30:
The best place to ask this would be the gitter channel. 

Essentially the hardware has nothing to do with this question - whether it arms or takes off depends on the flight stack - usually ArduPilot or PX4. Which one are you using? You should put that on the gitter chat.

Try to take off and arm using a GCS (say QGroundControl). If that works, then you know that there is no hardware issue. 

What you will need to do is run this code: http://python.dronekit.io/guide/taking_off.html
If it doesn't take off, then add some of the debugging information in the section that has the text "If you need more status information you"

Bashir1998, 2018-08-30:
Thank you very much for helping me .
I’ve Tried both ArduPilot and px4 .i have tried that code it connects to the Pixhawk and then it says waiting the vehicle to initialize. Which is in the code “ while not armble : print “waiting the vehicle to initialize “” and nothing happens it doesn’t takeoff.

hamishwillee, 2018-08-30:
So if it is waiting for the vehicle to initialise that is a very bad sign - essentially the flight stack is not even properly started. Suggest you use QGroundControl to add new firmware and try again.

hamishwillee, 2018-08-30:
Try with ArduPilot first - it is more suited to DK python.

Bashir1998, 2018-08-31:
Thanks alot i Will Try that 👍🏻

borhanreo, 2019-06-24:
Hi @Bashir1998 did you solved your problem??. if not then you can contact with me. i successfully connected GSM 3G/LTE between Raspberry PI and Ground station. **borhan.u.cse@gmail.com**