dronekit/dronekit-python #850: Cannot connect to the vehicle via telemetry

Rainerino, 2018-08-31:
I cannot connect to pixhawk 2.4.8 through telemetry. 
Using: 
Ubuntu 16.04 LTS, 
dronekit 2.9.1
pymavlink 2.2.10
pyserial 3.4

I am using a 915MHz telemetry and have tested it in Mission planner and QGroundControl with the exact same set up, have done multiple auto mission with it and haven't gotten any problems. 

I have tried: 
`v= dronekit.connect("/dev/ttyUSB0", baud=57600, rate=10, wait_ready=True)`

`v = dronekit.connect("/dev/ttyUSB0", baud=57600, wait_ready=False)`

Rainerino, 2018-09-08:
Do dreonkit development team ignore any problems in python 3 platform because it's not officially supported?

hamishwillee, 2018-09-09:
The team will take PRs for fixes in Python 3, but if you just raise an issue, then the chances of it being fixed in any sort of timely manner are low.

Rainerino, 2018-09-09:
Thank you @hamishwillee for the reply! I am not in a hurry, just want to fix this if possible.

hamishwillee, 2018-09-10:
@Rainerino Perhaps I should say "probably not fixed until someone creates a PR". All of the Python 3 work was done by community, and indeed most of the work over the last few years. There is some work done by the various projects that use this software, but not a great deal.

closed 2018-09-12 by Rainerino

Guohao91, 2020-01-09:
@Rainerino Hi, Yan yiyi? We have a extremely similar problem, as like connecting Pixhawk 4 with Companion computer via TELEM2 or micro USB Port.