dronekit/dronekit-python #853: Sudo python setup.py install ERROR

gbaweja, 2018-10-02:
`Processing lxml-4.2.5.tar.gz
Writing /tmp/easy_install-A8LXBh/lxml-4.2.5/setup.cfg
Running lxml-4.2.5/setup.py -q bdist_egg --dist-dir /tmp/easy_install-A8LXBh/lxml-4.2.5/egg-dist-tmp-GP1KKj
Building lxml version 4.2.5.
Building without Cython.
ERROR: /bin/sh: 1: xslt-config: not found

** make sure the development packages of libxml2 and libxslt are installed **

Using build configuration of libxslt 
In file included from src/lxml/etree.c:662:0:
src/lxml/includes/etree_defs.h:14:31: fatal error: libxml/xmlversion.h: No such file or directory
compilation terminated.
Compile failed: command 'x86_64-linux-gnu-gcc' failed with exit status 1
/tmp/easy_install-A8LXBh/lxml-4.2.5/temp/xmlXPathInitjoY7Gs.c:1:26: fatal error: libxml/xpath.h: No such file or directory
compilation terminated.
*********************************************************************************
Could not find function xmlCheckVersion in library libxml2. Is libxml2 installed?
*********************************************************************************
error: Setup script exited with error: command 'x86_64-linux-gnu-gcc' failed with exit status 1`

M0unir, 2018-10-11:
Seems you're missing a few libraries. You should do
```
sudo apt-get install python-pip python-dev python-lxml libxml2-dev libxslt-dev
```
Make sure you have all of those installed before trying again

gbaweja, 2018-10-12:
Thank you for your help. With your help i was able to solve it.

pietrodn, 2018-11-03:
Can this be closed?