dronekit/dronekit-python #854: RC Override for visioin positioning?

minominium, 2018-10-04:
Hey all. I want to ask some question before I'm dead exhausting...

I've seen several pages like http://python.dronekit.io/examples/channel_overrides.html or http://wiki.ros.org/mavros

What I want to do is drone to fly following some colored circles with OpenCV, in approximately 1m high.

First I tried setpoint topics, but after some research, I've noticed that there're some topics called 'RC Override'.

So, what I'm going to say, is...which is better solution for drone positioning with computer vision? setpoint topic? OR RC Override??

Any of your reply will help(or LITERALLY SAVE) me. Have a good time!

hamishwillee, 2018-10-04:
This isn't really a support channel, and nor is this  really a DK-Python question. Suggest you try on either the DK gitter channel or on the autopilot support channel for your autopilot of choice.

An example that does something like what you want is here: https://github.com/rmackay9/ardupilot-balloon-finder

closed 2018-10-04 by hamishwillee

minominium, 2018-10-05:
@hamishwillee whuw I'm sorry for my mistake. 12 hours ago is a.m.3 o'clock here in Asia :D

I  think I opened this issue in half sleepwalking situation.

Again, I'm sorry, and thanks for your work.