dronekit/dronekit-python #856: Drone control Web app development with python

MuhammadFaizanKhan, 2018-10-14:
I found that you have said
"DroneKit-Python allows developers to create apps that run on an onboard companion computer and communicate with the ArduPilot flight controller using a low-latency link. "
My question is that can i develop a web app with this? I mean you said the app will run on companion computer it means it will not run on a dedicated server?

hamishwillee, 2018-10-14:
There is nothing to stop you running DK-Python on a server instead of on a companion or a Ground Station. The only "issue" is making sure that your vehicle has a connection to the server.

There is a very simple example of this using CherryPy here http://python.dronekit.io/examples/drone_delivery.html

closed 2018-10-14 by hamishwillee

MuhammadFaizanKhan, 2018-10-15:
Thanks @hamishwillee for the reply, i guess it will work with all the supported drones which are support by the dronekit isnt? I want to test a general web app that support all those drones!

MuhammadFaizanKhan, 2018-10-20:
@hamishwillee When connecting with Vehicle through dedicated server, i should provide an IP address of the drone in the connection. One public drone IP should be require to work but if the user has multiple drone then how the connection will be made. 
Actually drones are connect with the GCs or my web app through the Mavlink protocol but which ip should be provide to the connection.? In my case the web app is running on the server and user want to connect its drone

hamishwillee, 2018-10-21:
My understanding is that each vehicle broadcasts on port 14550. So you would need from the web server to listen on `"whatever the ip address of each drone is":14550`. YOu'd need to create multiple vehicle objects and multiple connections. 

I haven't tried this and AFAIK there are no tests, so can't really comment "canonically" @peterbarker Anything you can add (like correcting my impression)?

@MuhammadFaizanKhan - This probably a good question to ask on the gitter channel. Active community there or people who are trying things.

MuhammadFaizanKhan, 2018-10-25:
Thanks @hamishwillee 

Each vehicle broadcast on **14550** port then, using a single IP address how can we make multiple connection object?? How the DK will differentiate the drone? You know, usually each user has a single IP address (Do the end user need to purchase IP for his each drone?). If the user assign this ip address to drone (If he can?? do it supported in ardupilot drone?) or multiple drones then how the DK will know that which drone should be interact? 

Further see the , initial the interaction mechanism look like this 
1. Drone Connected to a Dedicated Server using (Mavlink Protocol)
2. The drone user connected to the browser accessing the dedicated server through internet in order to interact with drone.

drone---->(connected via mavlink)--->(dedicated server running DK)
                                                                        ^
                                                                         |
                                                                  internet
                                                                        ^
                                                                         |
                                                                End user with the browser
Is this valid? The drone will communicate with the dedicated server via mavlink protocol and user accessign DK through the internet and controlling the drone.

I will happy if @peterbarker  provide some clarification about this matter.

hamishwillee, 2018-10-25:
> @MuhammadFaizanKhan - This probably a good question to ask on the gitter channel. Active community there or people who are trying things.

I haven't tried this and I don't know. Best to ask on the gitter channel.
Note though that MAVLink allows you to differentiate multiple vehicles on a port using the system ID. This may require that you use a slightly different version of DroneKit because the underlying libraries don't properly support system id filtering.

MuhammadFaizanKhan, 2018-10-29:
Remember the thread is still open!! I want explanation that, is this possible to send MavLink commands to the dronekit (vice versa) using TCP/IP connection? means over the internet.

hamishwillee, 2018-10-29:
No it isn't:
[image]

I'm happy for this to be reopened, but I'm at the end of my knowledge, so you may find yourself talking to an empty room.

Yes it is possible to send commands using TCP or UDP protocols over the Internet - provided you know the host address. BUT as far as I know TCP will only work for ArduPilot (not supported for PX4)

The gitter link is: https://gitter.im/dronekit/dronekit-python