dronekit/dronekit-python #858: Why choose dronekit over droneCode

MuhammadFaizanKhan, 2018-10-18:
3D robotics is behind both projects (dronekit and dronecode). I am searching for such an sdk that support most of th drones. I found that both sdks are same then, why should i choose dronekit over dronecode?

hamishwillee, 2018-10-18:
The situation is not quite as it appears in the docs. 3DR does not directly support any of the development for either library.

Dronecode "owns" both projects. It is only actively developing the Dronecode SDK. This is still in "beta"/near candidate release state for the core C++ library, and while Python and Swift libraries are well underway, there is no "firm" ETA. This is optimised and tested for PX4. It *anecdotally* works with ArduPilot, but no testing is done to make sure that is the case. Areas of weakness with ardupilot likely to be in Mission upload/download and Parameters.

DroneKit is governed by Dronecode, but there is no real resource behind it. The Python API has an active community on the gitter channel. It is optimised for ArduPilot, but does also work for PX4. It is really easy to hack and is essentially released. However if you have a problem that you want to fix in the core, you'll need to roll it in yourself. 

If you're working with PX4 then the Dronecode SDK is definitely going to be the future. If you know C++ it can be used now quite effectively. 
If you're working with ArduPilot or you can only use Python then DroneKit Python is your best choice (though longer term I hope that the SDK will support both effectively).

Another option to consider if you're working on Android is something like https://blog.adafruit.com/2018/05/04/rosetta-drone-open-source-android-based-mavlink-wrapper-for-dji-drones-diu_x-dronecode-djiglobal/ , which provides an interface layer for MAVLink translation.

closed 2018-10-18 by hamishwillee

MuhammadFaizanKhan, 2018-10-19:
Thanks, but as I told you before that i am searching for the SDK that work with most of the drones. Do you have any idea about it (cause you are only talking about PX4 and Ardupilot)?

Further, why there is emphasize on **Ardupilot** and **PX4** only. I understand that Dronekit or dronecode both will be compatible with **Mavlink** supported vehicles (It should work with all vehicles as it tells us). Then, why emphasizing only on Ardupilot and PX4 autopilots?. Do both Autopilot work differently in the SDK? and what about the any third autopilot that support mavlink protocol (like parrot) or not (like DJI).

amilcarlucas, 2018-10-19:
dronekit talks mavlink. A lot of autopilots talk mavlink. Hence dronekit can control a lot of autopilots.

But most people here use either Ardupilot and/or PX4 so we can only talk about systems that we know and use.

But yes, it should work for any mavlink autopilot.

hamishwillee, 2018-10-20:
As @amilcarlucas says - I'm talking about the autopilots I know. A few are listed here: https://mavlink.io/en/about/implementations.html
There are also many vehicles that use MAVLink that don't advertise their autopilots. 

At the lowest level MAVLink is an over the wire message standard. That stuff works reliably on any system that supports the standard. There are higher level protocols layered over the messaging standard to provide specific services - like uploading missions etc. That stuff is not as tightly specified and there may/are minor differences in implementations. Something like DroneKit compensates for those minor differences for ArduPilot and PX4 ... but I don't know about how well it does so for other sytems.

hamishwillee, 2018-10-20:
PS There is no "rosetta system" that every drone uses for offboard comms. DJI for example do not use MAVLink, which is why I pointed to that rosetta app for android that translates MAVLink to  DJI and back.

joshferns, 2018-10-26:
Hello,
Could someone please explain how does ROS+ mavros fit in when compared to dronekit?
Are these competing platforms or are they somehow complementary?

amilcarlucas, 2018-10-26:
competing. They do the same thing.

joshferns, 2018-10-26:
Is dronekit somehow easier? I'm trying to figure out which one to take up, and haven't been able to find a comparison between the two.
For context, I plan to make a swarm of drones connected to a ground station, have already made some progress using UGCS on windows, but want to switch over to using an OBC.

hamishwillee, 2018-10-28:
ROS is a generic robotics toolkit. It comes with lots of useful higher level features that are independent of the flight stack - e.g. like tools for computer vision. But there is a significant learning curve there and a lot more overhead. Also the simulator that comes with it is linux only (don't know about ROS itself).

DroneKit is IMO much easier for the easy stuff.

MuhammadFaizanKhan, 2018-10-29:
Yes, ROS is a generic robotics toolkit that also cover drones  while dronekit is specifically designed for mavlink supported drones. **Maybe** ROS provide you large number of drones compare to dronekit!