dronekit/dronekit-python #859: dronekit.APIException: Timeout in initializing connection

gbaweja, 2018-10-18:
I am using Intel UpBoard with Pixhwak Cube. Pixhawk cube is connected with UpBoard via FTDI Cable, on the USB port of UpBoard and Telemetry 2 on Pixhawk.

I am running the command..

`python wayPoint1.py --connect "/dev/ttyUSB0"`

and i am getting this error.

```
Connecting to vehicle on: /dev/ttyUSB0
>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "wayPoint1.py", line 24, in <module>
    vehicle = connect(args.connect, baud=57600, wait_ready=True)
  File "build/bdist.linux-x86_64/egg/dronekit/__init__.py", line 3018, in connect
  File "build/bdist.linux-x86_64/egg/dronekit/__init__.py", line 2262, in initialize
dronekit.APIException: Timeout in initializing connection.
```

The wiring is correct, and the firmware i am using on Pixhawk is ArduCopter. Please suggest where am i going wrong?

hamishwillee, 2018-10-18:
That message just means that dronekit python has not received a heartbeat. Two possibilities - it is connecting to the wrong port or the messages are not being sent. 

Messages are being sent if PX4 is set up correctly.
What version of PX4 are you using? If it is very recent you might want to check that the params for telemetry are set up properly as per https://docs.px4.io/en/peripherals/mavlink_peripherals.html
If it is a bit older (v1.7 say) check SYS_COMPANION parameter is set correctly. 

On the companion side `/dev/ttyUSB0` is what the documents say for the USB port. It is possible there are mulitple USB ports, so you might have another handle than that. I would check PX4 is working correctly first though.

gbaweja, 2018-10-20:
Thank you for your help. I want to let you know I have seen your comments on many blogs I know..I am a big fan of yours..you taking out your time and solving others problems..thats very appreciable and great..THANK YOU!!!

/dev/ttyUSB0 is the correct one I was using else it won't read the port and it will throw and error when the script will run. So I knew I am using the right port, but then I checked PX4.

I had connected PX4 via mini USB port to Mission Planner and then from a different laptop I ran the code to get connected to PX4 and it got connected. There was a pop window on Mission Planner that I had hit ok, I forgot what was the pop window, but that pop window came up when I connected the PX4 with mini USB port.

It was all working, but now my friend broke the mini USB port on the PX4, I am going to test the other PX4 that we have now. And I will update you to let you know if thats working.

Once again thank you for your help.

hamishwillee, 2018-10-21:
Thanks. OK, so mostly likely from this the problem is that PX4 is not sending the packets. 

@peterbarker Dronekit on UpBoard connected to Cube via FTDI Cable from UpBoard USB port to Cube TELEM2. Cube is running ArduCopter. Connection is "/dev/ttyUSB0" but connection is timing out. They've had a connection to mission planner via this same connection, so my guess is ArduCopter end not sending mavlink packets. Any advice on debugging?

peterbarker, 2018-10-22:
On Sun, 21 Oct 2018, Hamish Willee wrote:

> @peterbarker Dronekit on UpBoard connected to Cube via FTDI Cable from
> UpBoard USB port to Cube TELEM2. Cube is running ArduCopter. Connection is
> "/dev/ttyUSB0" but connection is timing out. They've [...]

Run mavproxy on the UP board.

gbaweja, 2019-02-14:
@hamishwillee @peterbarker 

I have a question related to the problem above, but if you think I should be opening different query for this question and if this place is not the right place, please let me know I will do it.

My question is,  Dronekit on UpBoard connected to Cube via FTDI Cable from UpBoard USB port to Cube TELEM2. Cube is running ArduCopter. Connection is "/dev/ttyUSB0" and connection is being established also, but it takes about around 3 minutes approximately to get connected and then the motors start moving.
Could you please tell me what is causing the time delay?And how can I solve this, because I do not want it to take 3-4 minutes to get connected. Please suggest.

hamishwillee, 2019-02-14:
@gbaweja I have no idea. I'd be asking on the gitter channel, to increase your chance of getting more people who have experience with this setup.

gbaweja, 2019-02-21:
I did that..thank you for your help..but I am not able to get any answer from anywhere. Do you think it has to do something with PX4. I had the same problem when I used Raspberry Pi as the companion computer. 

The issue is: It just takes about 2 -3 minutes to establish the connection with cube after I run the drone kit script simple_goto.py..whether I am running on raspberry pi or intel upboard.

peterbarker, 2019-03-13:
On Thu, 14 Feb 2019, gbaweja wrote:

> I have a question related to the problem above, but if you think I should be opening different query for this question and if this place is not the right
> place, please let me know I will do it.
>
> My [...]

Which script were you running that had the motors start to move?

Guohao91, 2020-01-03:
@gbaweja we have the similar problem now. What my configurations are: Pixhawk 4(installed PX4), Intel upboard. and my software is: 
dronekit                      2.9.2 
pymavlink                     2.4.3
MAVProxy                      1.6.2.
I use micro USB port to connect companion computer, the problem shows:
[image]
@peterbarker  @hamishwillee  Please help!

Guohao91, 2020-01-03:
I also tried TELEM2 port, it doesn't work fine either. the problem is:
 
[image]

gbaweja, 2020-01-13:
@Guohao91 I can share with you what solved my problem and looks like you might also have the same solution to it as per what you have shared here.

The versions of Dronekit, Pymavlink and Mavproxy are critically important. I had the wrong versions and I changed the versions to solve my problem.

Try this, I am pretty sure it will solve your problem.

`pip install dronekit==2.9.1`
`pip install Pymavlinl==2.2.10`
`pip install Mavproxy==1.6.4`

Install Mavproxy at the end. You may run into lxml library issues if you already don't have. All you would have to do in that case install lxml libraries and its dependencies.

Guohao91, 2020-01-14:
@gbaweja 
Appreciate it！I'll try on later, thanks a lot!

Guohao91, 2020-01-14:
@gbaweja 
hi, I need to check to make sure: if you were using PX4? and then come up with dronekit codes? I been told MavSDK should work with PX4, while dronekit is hardcoded to Ardupilot......

gbaweja, 2020-01-20:
@peterbarker @hamishwillee So I am again having this issue. 

Hardware: 
                   Raspberry Pi to Pixhawk
                  Raspberry Pi GPIO PINS connected to Pixhawk Cube Mini on Telemetry 2.
                  RC is on GUIDED MODE

Software:
                  Dronekit==2.9.1
                 Pymavlinl==2.2.10
                 Mavproxy==1.6.4
    
                All these versions are available under `sudo pip list` and `pip list`

I also ran `mavproxy.py --master=/dev/ttyS0 --baudrate 921600 --aircraft MyCopte` it did run the motors but I am still having the issue with Dronekit Timout.

Guohao91, 2020-01-21:
@gbaweja try to use another serial port. like micro-usb port.

harryhancock, 2022-01-28:
@gbaweja did you manage to fix the timeout issue using USB0? I'm having the same issue with  an FTDI cable but using Raspberry Pi