dronekit/dronekit-python #862: How to MUTE connection logs?

mustaphos, 2018-10-25:
Hi.
Is there a way to mute console log outputs?
Thanks for answers.

pietrodn, 2018-11-01:
You could monkey-patch the methods in `dronekit/util.py`:

```python
from dronekit import util
util.errprinter = lambda msg: None
util.logger = lambda msg: None
```

Caveat: do this before importing the DroneKit module.

pietrodn, 2018-12-29:
Now dronekit uses the built-in Python `logging` module. Now you can do:

```python
import logging
logging.getLogger('dronekit').addHandler(logging.NullHandler())
logging.getLogger('autopilot').addHandler(logging.NullHandler())
```

Where the `autopilot` logger handles the log messages coming from the firmware, while the `dronekit` logger handles the messages coming from the library.

I think this issue can be closed now.

mustaphos, 2019-01-06:
@pietrodn thanks for your support.

closed 2019-01-06 by mustaphos