dronekit/dronekit-python #863: High latency in MAVLink message receiving

pietrodn, 2018-10-27:
I wrote a simple script to measure the MAVLink message latency (round-trip time) by exploiting the `TIMESYNC` protocol as implemented in PX4.
The protocol works like this: in the client application you send a `TIMESYNC` message to the autopilot, with:
* `tc1=0`;
* `ts1` set to the locally generated timestamp.

The firmware responds ([source code](https://github.com/PX4/Firmware/blob/6f204c491335af88dcf1c13484e3d98f82868296/src/modules/mavlink/mavlink_timesync.cpp#L67)) with another `TIMESYNC`, having:
* the same `ts1` field (the timestamp of the original message);
* the autopilot's timestamp in the `tc1` field.

By observing the difference between the timestamp of arrival of the second message and its `ts1` field, you can effectively measure the round-trip time of the message and estimate the link latency (e.g. by assuming latency = RTT / 2).
You can also estimate the offset between the client and the autopilot's clocks using the `tc1` field, but I'm not interested in this right now.

I did the following experiment: send a `TIMESYNC` to the autopilot every 10 ms for 100 times, and measure the round-trip times from the answers.
I'm using the PX4 SITL simulator, so I'm communicating with the autopilot through UDP on the same host (ideal conditions, no packet loss).

I noticed a strange pattern in the measured latencies (**orange line** in the graph).

[image]

In the graph we can observe groups of 4-5 messages with strictly decreasing latencies, starting from 25 ms  down to about 5 ms or less.

I also observed the autopilot-generated timestamps (`tc1` field) and noticed that they were very regular (there were about 10 ms between them, always).
So, this must be a problem on the receiving side of the client.

The cause of the problem is [this line of code](https://github.com/dronekit/dronekit-python/blob/5d2614f79092cc3ea03062a084579eb16577cd22/dronekit/mavlink.py#L207): `time.sleep(0.05)` in the `mavlink_thread_in` function.

That `sleep()` makes DroneKit wait 50 ms if no messages are found in the buffer; it matches the 25 ms delay shown in the graph (I'm plotting RTT / 2 as a latency estimate).

By removing the sleep, I get dramatically lower latencies (less than 1 ms on average): they are shown on the **blue line** in the above graph.

Why is that sleep there? Is it possible to remove it?
I'd like to do so to obtain a reliable estimate  of the link latency on the autopilot side.

pietrodn, 2018-10-31:
Fixed by #864.

closed 2018-10-31 by pietrodn

dwiel, 2020-06-02:
I'm seeing this exact same pattern on latest stable dronekit and arducopter SITL. Any ideas where else a similar issue might be cropping up? I am connecting to the SITL directly over TCP (no mavproxy). The maximum round trip time I am seeing is around 36 ms. I can supply additional data/plots/code tomorrow. Would anything in particular be helpful?

pietrodn, 2020-06-02:
Can you please provide the code you’re running, the latency plots, the Python version and the DroneKit version? That could help 😊

dwiel, 2020-06-02:
Sorry it took me a minute to remove the proprietary code from my experiments. Hopefully this will help reproduce:

```
import sys
import time

import dronekit
import numpy as np
import os

latencies = []
last_vehicle_time = 0

def timesync_listener(vehicle, name, message):
    global last_vehicle_time

    received_time = int(time.time() * 1e6)

    latency = (received_time - message.ts1) / 1e3
    if latency > 100:
        return

    print("vehicle time:", message.tc1)
    print("vehicle time delta:", (message.tc1 - last_vehicle_time) / 1e3)
    last_vehicle_time = message.tc1

    print(f"latency in ms: {latency:.02f}")
    latencies.append(latency)

def heartbeat():
    vehicle = dronekit.connect("tcp:localhost:6769")

    vehicle.add_message_listener("TIMESYNC", timesync_listener)

    fps = 100
    # rate = Rate(fps)
    for _ in range(5 * fps):
        vehicle.send_mavlink(
            vehicle.message_factory.timesync_encode(0, int(time.time() * 1e6))
        )

        # rate.sleep()
        time.sleep(1 / fps)

    # wait for all of the timesync messages to return
    time.sleep(1)

    print(latencies)
    print(f"mean: {np.mean(latencies):.03f} std: {np.std(latencies):.03f}")
    print(f"max: {np.max(latencies):.03f} min: {np.min(latencies):.03f}")
    print(f"python version: {sys.version}")
    os.system(f"pip3 list | grep dronekit")

if __name__ == "__main__":
    heartbeat()

```
and the abbreviated output:
```
[34.138, 23.513, 13.457, 3.415, 26.83, 16.794, 6.509, 30.285, 20.02, 9.737, 11.998, 21.93, 11.856, 1.767, 25.413, 15.33, 5.257, 29.067, 18.949, 8.881, 21.91, 11.848, 1.789, 25.44, 15.375, 5.277, 28.556, 18.445, 8.278, 24.642, 14.606, 4.52, 27.527, 17.42, 7.304, 30.491, 20.395, 10.292, 0.281, 23.271, 13.173, 3.073, 26.283, 16.18, 6.074, 29.311, 19.212, 9.109, 32.283, 22.21, 12.115, 2.02, 24.847, 14.757, 4.66, 27.72, 17.628, 7.351, 30.529, 20.44, 10.373, 0.339, 22.98, 12.881, 2.622, 25.734, 15.651, 5.585, 28.486, 18.415, 8.32, 31.146, 21.085, 10.992, 0.928, 23.703, 13.685, 3.645, 26.39, 16.286, 6.173, 29.007, 18.898, 8.816, 31.715, 21.628, 11.517, 1.406, 24.381, 14.248, 4.137, 27.016, 16.756, 6.663, 29.254, 19.143, 9.037, 31.855, 21.745, 11.635, 1.531, 24.375, 14.228, 4.114, 26.74, 16.615, 6.504, 29.344, 19.333, 9.323, 32.053, 22.046, 12.051, 2.055, 25.003, 14.918, 4.873, 27.262, 17.204, 7.113, 29.493, 19.435, 9.355, 31.668, 21.577, 11.479, 1.382, 23.717, 13.645, 3.547, 25.892, 15.801, 5.707, 28.145, 18.049, 7.946, 30.488, 20.402, 10.301, 0.27, 22.492, 12.383, 2.277, 24.846, 14.748, 4.642, 27.311, 17.211, 7.109, 29.708, 19.61, 9.509, 32.113, 22.016, 11.914, 1.81, 24.497, 14.401, 4.297, 27.074, 16.977, 6.916, 29.747, 19.483, 9.389, 32.199, 22.106, 12.019, 1.914, 24.63, 14.559, 4.467, 27.275, 17.181, 7.081, 29.958, 19.863, 9.779, 0.295, 22.682, 12.592, 2.493, 25.41, 15.315, 5.202, 28.211, 18.114, 8.015, 31.092, 20.999, 10.887, 0.783, 23.731, 13.645, 3.555, 26.568, 16.47, 6.365, 29.593, 19.547, 9.467, 32.683, 22.445, 12.383, 2.282, 25.344, 15.266, 5.19, 28.605, 18.514, 8.43, 31.673, 21.613, 11.534, 1.454, 24.557, 14.489, 4.397, 27.819, 17.732, 7.631, 30.874, 20.785, 10.715, 0.636, 23.911, 13.818, 3.716, 27.003, 16.917, 6.807, 30.042, 19.937, 9.908, 1.826, 23.37, 13.303, 3.055, 26.686, 16.595, 6.523, 29.869, 19.756, 9.652, 33.114, 23.033, 12.979, 2.898, 26.404, 16.307, 6.206, 29.653, 19.567, 9.488, 32.953, 22.871, 12.759, 2.668, 26.016, 15.932, 5.846, 29.176, 19.092, 8.987, 32.259, 22.165, 12.06, 1.969, 25.233, 15.161, 5.098, 28.322, 18.234, 8.135, 31.318, 21.108, 10.986, 0.879, 24.141, 14.046, 3.942, 27.139, 16.795, 6.53, 29.835, 19.755, 9.676, 32.795, 22.725, 12.662, 2.575, 25.545, 15.442, 5.339, 28.366, 18.26, 8.152, 31.024, 20.929, 10.825, 0.745, 23.703, 13.618, 3.382, 26.274, 16.186, 6.092, 29.033, 19.006, 8.925, 31.778, 21.704, 11.624, 1.531, 24.212, 14.151, 4.059, 26.747, 16.703, 6.612, 29.109, 19.058, 8.938, 31.552, 21.32, 11.248, 1.145, 23.827, 13.731, 3.636, 26.246, 16.154, 6.06, 28.726, 18.64, 8.553, 31.284, 21.214, 11.172, 0.921, 23.434, 13.347, 3.26, 25.52, 15.489, 5.247, 27.58, 17.504, 7.398, 29.824, 19.722, 9.632, 32.103, 21.997, 11.902, 1.808, 24.468, 14.417, 4.348, 26.957, 16.893, 6.691, 29.387, 19.086, 8.876, 31.549, 21.278, 11.173, 1.081, 23.665, 13.532, 3.457, 26.073, 16.017, 5.93, 28.709, 18.606, 8.504, 31.352, 21.311, 11.222, 1.129, 23.819, 13.733, 3.647, 26.409, 16.361, 6.242, 29.145, 19.018, 8.918, 31.803, 21.7, 11.591, 1.483, 24.437, 14.332, 4.235, 27.307, 17.206, 7.096, 29.934, 19.67, 9.555, 32.664, 22.572, 12.483, 2.419, 25.543, 15.453, 5.307, 28.235, 18.135, 8.028, 30.989, 20.887, 10.784, 0.676, 23.998, 13.92, 3.841, 26.908, 16.814, 6.761, 29.994, 19.882, 9.792, 0.342, 23.002, 12.949, 2.857, 26.137, 16.098, 6.035, 29.463, 19.387, 9.142, 32.572, 22.424, 12.153, 2.043, 25.323, 15.248, 5.143, 28.174, 18.074, 7.971, 31.324, 21.193, 11.083, 0.975, 24.151, 14.05, 3.943, 27.027, 16.931, 6.836, 29.988, 19.914, 9.826, 32.881, 22.779, 12.668, 2.565, 25.643, 15.544, 5.434, 28.373, 18.264, 8.177, 31.199, 21.098, 10.993, 0.888, 23.744, 13.665, 3.542, 26.597, 16.5, 6.388, 29.37, 19.269, 9.189, 32.088, 21.989, 11.88, 1.771, 24.651, 14.398, 4.309, 27.138, 17.072, 6.984, 29.719, 19.622, 9.514, 32.448, 22.355, 12.269, 2.203]
mean: 16.467 std: 9.502
max: 34.138 min: 0.270
python version: 3.6.8 (default, Oct  7 2019, 12:59:55)
[GCC 8.3.0]
dronekit (2.9.2)
dronekit-sitl (3.3.0)
```
and plots of the first 100 steps (1 second) of the data pasted above:

[image]

dwiel, 2020-06-02:
it looks like this is probably specific to arducopter/SITL. Here is a plot of 10 seconds of round trip times in milliseconds measured at 100hz on an actual hardware flight controller:

[image]

dwiel, 2020-06-02:
and here is the first second from the plot above:

[image]

Definitely some similar patterns, but not quite as stark

pietrodn, 2020-06-02:
Hello, I found in my computer the script that I originally used to run the first analysis.
Can you try running it to see if it returns the same results? It's here: https://gist.github.com/pietrodn/85833adb5c4caf46fade82a35a30fa58

dwiel, 2020-06-03:
that said, it's looking to me like the most likely causes actually in the arducopter SITL. I do not have this problem when communicating with my hardware flight controller:

```
Sending 30 pings to vehicle...
TIMESYNC seq=0 time=10.509 ms
TIMESYNC seq=1 time=6.244 ms
TIMESYNC seq=2 time=8.726 ms
TIMESYNC seq=3 time=5.030 ms
TIMESYNC seq=4 time=5.349 ms
TIMESYNC seq=5 time=4.123 ms
TIMESYNC seq=6 time=6.990 ms
TIMESYNC seq=7 time=16.980 ms
TIMESYNC seq=8 time=9.326 ms
TIMESYNC seq=9 time=7.272 ms
TIMESYNC seq=10 time=4.795 ms
TIMESYNC seq=11 time=7.767 ms
TIMESYNC seq=12 time=7.411 ms
TIMESYNC seq=13 time=6.289 ms
TIMESYNC seq=14 time=6.327 ms
TIMESYNC seq=15 time=6.968 ms
TIMESYNC seq=16 time=6.566 ms
TIMESYNC seq=17 time=7.421 ms
TIMESYNC seq=18 time=19.408 ms
TIMESYNC seq=19 time=21.411 ms
TIMESYNC seq=20 time=15.466 ms
TIMESYNC seq=21 time=6.308 ms
TIMESYNC seq=22 time=5.498 ms
TIMESYNC seq=23 time=4.743 ms
TIMESYNC seq=24 time=5.128 ms
TIMESYNC seq=25 time=4.699 ms
TIMESYNC seq=26 time=8.732 ms
TIMESYNC seq=27 time=14.459 ms
TIMESYNC seq=28 time=7.377 ms
TIMESYNC seq=29 time=4.631 ms
round-trip min/avg/max/stddev = 2.061/4.199/10.706/2.233
```

pietrodn, 2020-06-03:
Uh, I think I found the same bug on `dronekit-sitl`. Can you try removing this `sleep()` from this file and see if the situation improves? https://github.com/dronekit/dronekit-sitl/blob/master/dronekit_sitl/forwarder.py#L124

pietrodn, 2020-06-03:
Replacing it with `self.master.select(0.05)` should also work.

dwiel, 2020-06-03:
good find. There must be something similar in the ardupilot sitl. I just double checked and I am not hitting that code path. It is uninstalled just to be sure, and I am still having the same problem. I can post a new issue over in that repo.

pietrodn, 2020-06-03:
Sorry, maybe I didn't understand... isn't `dronekit-sitl` the ArduPilot SITL itself?

dwiel, 2020-06-03:
Just did a bit more research, and I see why there is confusion. "DroneKit-SITL is the fastest and easiest way to run SITL on Windows, Linux, or MAC OSX." However i'm running it using the method documented in the [ardupilot documentation](https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html).

dwiel, 2020-06-03:
Specifically, I am running the following command: `/ardupilot/build/sitl/bin/arducopter -S --model hexa --speedup 1 --defaults /ardupilot/Tools/autotest/default_params/copter.parm,/ardupilot/Tools/autotest/default_params/copter-hexa.parm --base-port 6769 -I0`

pietrodn, 2020-06-03:
Ok, clear. Unfortunately I'm not familiar with the internals of ArduPilot, I was a PX4 guy... 😁 
@peterbarker do you have any clue about the latency issue? Maybe it's a known problem or a regression?