dronekit/dronekit-python #872: Old Dependencies (future, pymavlink)

montge, 2018-11-05:
DroneKit appears to have some older dependencies that need updating.

dronekit 2.9.1 has requirement future==0.15.2, but you'll have future 0.17.1 which is incompatible.
dronekit 2.9.1 has requirement pymavlink==2.0.6, but you'll have pymavlink 2.2.21 which is incompatible.

Unfortunately other libraries we're using require newer versions.

montge, 2018-11-05:
Appears master is up to date, so may be that pip and anaconda need to incorporate changes.

pietrodn, 2018-11-05:
Uhm... there are still pinned dependencies in `requirements.txt` and `setup.py` which need fixing.
The problem also appears in CI. I think it's because the installation procedure installs `dronekit-sitl`, which in turn installs the old PyPI version of `dronekit`. This will be resolved when the next release is made.

pietrodn, 2018-11-07:
I think this can be closed, those errors will go away upon a new DroneKit release on PyPI.

mrpollo, 2018-11-13:
@peterbarker perhaps its best we cut a new release was this in your plans? do you have pypi.org credentials for dronekit?

pietrodn, 2019-01-08:
#894 could be beneficial for this issue, since it cuts the dependency link to `future`, which is not a direct `dronekit` dependency (no `import future` is present in the codebase).