dronekit/dronekit-python #876: How to operate ardupilot by position information by VIVE tracker

otsuka-kohei, 2018-11-16:
Hello.
I am drone begginer.

I am trying to hover a  pixhawk px4 drone at some position in room.
VIVE tracker provides high precision position data.
I would like to set target position and actual position (which is provided VIVE tracker) to px4, and hover px4 that target position automaticaly.

I found a solution that uses vehicle#channels#overrides method and my hand made  PID control code.
But it looks like danger way.

Could you tell me better solution for px4 automatic hoverring with VIVE tracker?

Thank you.

peterbarker, 2018-11-18:
On Thu, 15 Nov 2018, Otsuka Kohei wrote:

> I am trying to hover a pixhawk px4 drone at some position in room.
> VIVE tracker provides high precision position data.
> I would like to set target position and actual position (which is provided VIVE [...]

Both PX4 and ArduPilot (the most popular of the autopilot systems using 
Arduilot) allow offboard control using set_position_target_local_ned and 
similar messages.

otsuka-kohei, 2018-11-19:
I would like to use a method like below.  
`simple_go_ up(power_percentage, duration_ms)`