dronekit/dronekit-python #881: How to use remaining channels?

otsuka-kohei, 2018-11-26:
Hello.

I am building a drone which used Pixhawk PX4.
PX4 has 8 connectors for ESC and quad-X drone need only 4 connectors.
[Channel Overrides guide page](http://python.dronekit.io/examples/channel_overrides.html) indicated that "The remaining channel values are configurable, and their purpose can be determined using the RCn_FUNCTION parameters".
I would like to use remaining 2 channnels (ch7, ch8) for control additional individual ESCs that connected free PX4's connectors.

Please tell me how to use remaining channnels for control additional individual ESC?

Thank you.

hamishwillee, 2018-11-26:
That text is ArduPilot specific. If you are using PX4 then that whole page might be incorrect (it isn't tested). Specifically it is possible that PX4 even ignores the underlying MAVLink command to set parameter values.

Best place to discuss this may be the dronekit gitter channel.

otsuka-kohei, 2018-11-26:
Thanyou your response.
I will try to submit same question to GITTER too.