dronekit/dronekit-python #885: I just started to explore the drone kit and have a problem in step "Hello Drone"

unow0517, 2018-12-09:
Hi, kind experts!
i copied and pasted the basic "Hello Drone" script in QuickStart , but had errors like this 

 File "C:\Users\YOONHO\Desktop\python\Drone\Hello.py", line 7, in <module>
    from dronekit import connect
  File "C:\Users\YOONHO\AppData\Local\Programs\Python\Python37\lib\site-packages\dronekit\__init__.py", line 2192
    await = set(types)

line 7 is

 from dronekit import connect, VehicleMode

and i have totally no idea about error in 'await=set(types)'

I ,a newbie, am looking forward to generous experts opinion!
Thank you!!

pietrodn, 2018-12-29:
Duplicate of #891, the problem is Python 3.7 specific and it has been fixed in #840.
You simply need to use DroneKit from the GitHub master branch and not from PyPI. The version installed with `pip install dronekit` is pretty old. :-)

ingframin, 2019-01-03:
Is it possible to update the version on PyPI?

hamishwillee, 2019-01-04:
Probably - @peterbarker / @mrpollo  - update PyPI?

But you don't need to - you can install it locally using this git repo.