dronekit/dronekit-python #887: simple goto example of dronekit-python with ArduSub v3.5.3 gives me error of mode 19 not available

marmik18, 2018-12-14:
I have connected Pixhawk directly to my linux computer overr sub and the serial port is /dev/ttyACM0
When i run the code as 
`python simplegoto.py --connect /dev/ttyACM0`

dronekit detects the firmware as ArduSub but fails to detect heartbeat.
`$ python hello.py --connect /dev/ttyACM0 
Connecting to vehicle on: /dev/ttyACM0
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> MYGCS: 255, heartbeat lost
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> ArduSub V3.5.3 (ad81760b)
>>> PX4: 8d505a02 NuttX: 1a99ba58
>>> PX4v2 00350045 33375111 31313639
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 19 not available on mavlink definition
`
Can anyone help me out with this is issue?
Thank you in advance.

jmachuca77, 2018-12-14:
Reinstall dronekit from the github repository instead of using pip

marmik18, 2018-12-15:
I got it solved, i updated pymavlink to 2.3 from 2.0 which was actually supported by dronekit

closed 2021-04-04 by marmik18