dronekit/dronekit-python #891: Cannot import dronekit on Python 3

otsuka-kohei, 2018-12-18:
Hello.

I am trying make programs with dronekit and python 3 on windows 10 1809.
But when I import dronekit library, python shows error like bellow.
```
>>> from dronekit import connect
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\otsuka\AppData\Local\Programs\Python\Python37-32\lib\site-packages\dronekit\__init__.py", line 2192
    await = set(types)
          ^
SyntaxError: invalid syntax
```

Must I use python 2.7?

Thank you.

peterbarker, 2018-12-18:
On Mon, 17 Dec 2018, Otsuka Kohei wrote:

> I am trying make programs with dronekit and python 3 on windows 10 1809.
> But when I import dronekit library, python shows error like bellow.

Are you using dronekit-python from the github repository or pipi?

If that latter - try the former.

otsuka-kohei, 2018-12-18:
Thank you for your response.

`pip list` result like bellow.
```
Package           Version
----------------- --------
astroid           2.1.0
colorama          0.4.1
dronekit          2.9.1
```
dronekit which I installed looks like latest release.

I installed dronekit by `pip install dronekit`.
If I install dronekit from github repo, I have to execute instruction according to [this page](http://python.dronekit.io/1.5.0/contributing/developer_setup_windows.html)?

otsuka-kohei, 2018-12-20:
I replaced from pip installed dronekit to latest github repo dronekit source, and SyntaxError does not appear.
Thank you.

But I stuck on #759, so I cannot use dronekit on Python 3.7.1 successfully.

pietrodn, 2018-12-29:
The "await" bug was fixed in #840. So you need to use master.
Try to enable detailed logging:
```
import logging
logging.basicConfig(level=logging.DEBUG)
```

at the beginning of your application, and please post a code sample and the full logs.

pietrodn, 2018-12-29:
I just tried and DroneKit's tests run successfully on Python 3.7: [CircleCI logs](https://circleci.com/gh/pietrodn/dronekit-python/148)

drnic, 2019-02-13:
For others, I added the following to my `requirements.txt`:

```
-e git://github.com/dronekit/dronekit-python#egg=dronekit
```

justinclift, 2020-07-02:
This seems like it could be closed, with the 2.9.2 release having been done a while back.

As a related data point though, the DroneKit installation page specifically says Python 3.6 is **not** supported:

&nbsp; &nbsp; https://dronekit-python.readthedocs.io/en/latest/develop/installation.html

Is that just a case of the docs needing to have the latest version deployed?  [That doc site](https://dronekit-python.readthedocs.io/en/latest/about/release_notes.html) is still saying 2.9.1 is the latest release.

pietrodn, 2020-07-02:
The docs need to be updated. [The tests are passing on Python 3.6](https://app.circleci.com/pipelines/github/dronekit/dronekit-python/jobs/1416), and I was literally flying drones with DroneKit on Python 3 :-)

pietrodn, 2020-07-02:
I made this PR: https://github.com/dronekit/dronekit-python/pull/1040

justinclift, 2020-07-02:
> ... I was literally flying drones with DroneKit on Python 3

Thanks @pietrodn. :smile:

Yeah, I was looking around for an ArduPilot SDK, but skipped DroneKit because the official docs **literally say** it's Python 2 only and Python 3 isn't supported.

Fortunately, a fellow project dev also mentioned they've used DroneKit with Python 3, thus took a second look. :smile:

Turbullance, 2020-11-02:
Hi how do i install dronekit using the github respository?

AnaghBenjwal, 2022-11-07:
> Hi how do i install dronekit using the github respository?

were you able to solve this?

AlexanderZ3, 2022-11-10:
hi

not yet
you can try Tello EDU  to validate the drone-related algorithm. which has mature SDK and is not expensive. 

best
zezhong