dronekit/dronekit-python #893: Re-enable AppVeyor CI

pietrodn, 2018-12-29:
After the fixes in #878, the AppVeyor configuration for CI is good to go.
However AppVeyor is not enabled for the repo.

I think it would be a good idea to enable it in order to have Windows testing too. :-)

peterbarker, 2018-12-30:
On Sat, 29 Dec 2018, Pietro De Nicolao wrote:

> After the fixes in #878, the AppVeyor configuration for CI is good to go.
> However AppVeyor is not enabled for the repo.

Well.. I changed the webhook to "active".  Let me know how it works out 
:-)

pietrodn, 2018-12-30:
Uhm, can you also force it to make a build of the current master? :-)

peterbarker, 2018-12-31:
On Sun, 30 Dec 2018, Pietro De Nicolao wrote:

> Uhm, can you also force it to make a build of the current master? :-)

Nope.  You'll have to get a nice PR together :-P

pietrodn, 2018-12-31:
The latest commit built successfully, so this issue is solved!

closed 2018-12-31 by pietrodn