dronekit/dronekit-python #900: support for RC_CHANNELS  message

tizianofiorenzani, 2019-01-24:
I noticed that when the SERIAL_PROTOCOL is set to 2 (Mavlink2), the RC_CHANNELS_RAW is not sent, and RC_CHANNELS  takes its place and dronekit cannot read the rc channels anymore.

tizianofiorenzani, 2019-01-24:
In the meantime in my script...

`            
    @vehicle.on_message('RC_CHANNELS')
    def listener(vehicle, name, message):
            def set_rc(chnum, v):
                vehicle._channels._update_channel(str(chnum), v)

            set_rc(1, message.chan1_raw)
            set_rc(2, message.chan2_raw)
            set_rc(3, message.chan3_raw)
            set_rc(4, message.chan4_raw)
            set_rc(5, message.chan5_raw)
            set_rc(6, message.chan6_raw)
            set_rc(7, message.chan7_raw)
            set_rc(8, message.chan8_raw)
            vehicle.notify_attribute_listeners('channels', vehicle.channels)   
`

agising, 2019-08-16:
Thank you very much, this helped me a lot! I did expand for channels 9-16 and it works the same.

`@vehicle.on_message('RC_CHANNELS')
	def RC_CHANNEL_listener(vehicle, name, message):
		# Channel 1-8 must be updated here in the simulation. Not required on real autopilot.
		set_rc(1, message.chan1_raw)
		set_rc(2, message.chan2_raw)
		set_rc(3, message.chan3_raw)
		set_rc(4, message.chan4_raw)
		set_rc(5, message.chan5_raw)
		set_rc(6, message.chan6_raw)
		set_rc(7, message.chan7_raw)
		set_rc(8, message.chan8_raw)
		set_rc(9, message.chan9_raw)
		set_rc(10, message.chan10_raw)
		set_rc(11, message.chan11_raw)
		set_rc(12, message.chan12_raw)
		set_rc(13, message.chan13_raw)
		set_rc(14, message.chan14_raw)
		set_rc(15, message.chan15_raw)
		set_rc(16, message.chan16_raw)
		vehicle.notify_attribute_listeners('channels', vehicle.channels)
	
	def set_rc(chnum, v):
		vehicle._channels._update_channel(str(chnum), v)
        

        print("Now I can read all my RC-channels, here is the value of ch13", vehicle.channels[13])
`
Thumbs up!

angela804, 2022-10-24:
I tried setting serial protocol to 1 and it works, though. I can read the channels