dronekit/dronekit-python #902: Obstacle Avoidance with HC-SR04 Ultrasonic Sensors and DK-PY

gauravsarraf, 2019-01-25:
I was wondering if there is any way I can use these cheap sensors for obstacle avoidance with a Raspberry Pi and Dronekit. I know this can be done relatively easily with the RC inputs first fed into a on-board Arduino UNO and then sending a MAVLink message every second. 

But, I also want to do some image recognition with RPi, therefore I am trying to incorporate both the Obstacle Avoidance and Image Recognition with RPi. It will save me space, battery consumption, excess weight of two boards on the drone and most importantly, a Telemetry Port on PIXHawk.

Thanks. :)

nagendra092918, 2019-03-04:
did u implemented this if yes,please explain or send the details of that to nagendra092918@gmail.com

SohamKorgaonkar, 2019-07-05:
I have created a GitHub repo just for that https://github.com/SohamKorgaonkar/object_avoidance_in_drones