dronekit/dronekit-python #907: Parameter name keys have extra characters

custeve, 2019-02-08:
Using the following code: 

`print("\nPrint all parameters (iterate `vehicle.parameters`):")
for key in vehicle.parameters:
    print(" Key:%s Value:%s" % `(key,vehicle.parameters[key])
'
The names have extra "b'(" prefixes and random \x00 suffixes. Example result snippet. This is running plane simulation in MissionPlanner and dronekit_sitl. 

' Key:b'RC13_MIN\x00\x00\x00\x00\x00\x00\x00\x00' Value:1100.0
 Key:b'RC13_TRIM\x00\x00\x00\x00\x00\x00\x00' Value:1500.0
 Key:b'RC13_MAX\x00\x00\x00\x00\x00\x00\x00\x00' Value:1900.0
 Key:b'RC13_REVERSED\x00\x00\x00' Value:0.0
 Key:b'RC13_DZ\x00\x00\x00\x00\x00\x00\x00\x00\x00' Value:0.0
 Key:b'RC13_OPTION\x00\x00\x00\x00\x00' Value:0.0
 Key:b'RC14_MIN\x00\x00\x00\x00\x00\x00\x00\x00' Value:1100.0
 Key:b'RC14_TRIM\x00\x00\x00\x00\x00\x00\x00' Value:1500.0
 Key:b'RC14_MAX\x00\x00\x00\x00\x00\x00\x00\x00' Value:1900.0
 Key:b'RC14_REVERSED\x00\x00\x00' Value:0.0
 Key:b'RC14_DZ\x00\x00\x00\x00\x00\x00\x00\x00\x00' Value:0.0
'

custeve, 2019-02-09:
Update, I have determined this is a porting issue from Python 2 to Python 3, with the handling of strings as binary or string. I have not yet determined the best resolution to the problem.

custeve, 2019-02-10:
Update: I was able to get the following to work: 

`vehicle.parameters[b'TRIM_ARSPD_CM\x00\x00\x00'] = 1755`
and
`vehicle.parameters.add_attribute_listener(b'TRIM_ARSPD_CM\x00\x00\x00', test_callback)`

However, it's not clear at this time the best long term fix, and whether the problem resides in DroneKit or pymavlink.

pietrodn, 2019-02-20:
Hi! Do you experience the problem using Python 2, Python 3, or both?

custeve, 2019-02-20:
Works fine in Python 2: 

```
print("\nPrint all parameters (iterate `vehicle.parameters`):")
for key in vehicle.parameters:
    print(" Key:%s Value:%s" % (key,vehicle.parameters[key]))
```

Snippet of return: 
```
Key:SERIAL4_BAUD Value:38.0
 Key:BATT2_MONITOR Value:0.0
 Key:TRIM_AUTO Value:0.0
 Key:RNGFND9_PIN Value:-1.0
 Key:RNGFND1_PIN Value:-1.0
 Key:STEER2SRV_DRTFCT Value:10.0
 Key:RC6_REVERSED Value:0.0
 Key:RC6_OPTION Value:0.0
 Key:RNGFND1_FUNCTION Value:0.0
 Key:SERVO5_REVERSED Value:0.0
 Key:INS_ACC2SCAL_Z Value:1.00100004673
 Key:SERVO4_REVERSED Value:0.0
 Key:INS_ACC2SCAL_X Value:1.00100004673
 Key:INS_ACC2SCAL_Y Value:1.00100004673
 Key:GPS_TYPE Value:1.0
 Key:RNGFND4_POS_X Value:0.0
 Key:RNGFND4_POS_Y Value:0.0
```

pietrodn, 2019-02-24:
I cannot reproduce the problem under Python 3.7.2 with the latest master.

I added as a SITL test your code:

```python
@with_sitl
def test_parameters_list(connpath):
    vehicle = connect(connpath)
    vehicle.wait_ready('parameters')
    print("\nPrint all parameters (iterate `vehicle.parameters`):")
    for key in vehicle.parameters:
        print(" Key:%s Value:%s" % (key, vehicle.parameters[key]))

    vehicle.close()
```

Here's what I get running it:

```
$ nosetests -svx dronekit.test.sitl.test_parameters:test_parameters_list

dronekit.test.sitl.test_parameters.test ...
Print all parameters (iterate `vehicle.parameters`):
 Key:SYSID_SW_MREV Value:120.0
 Key:SYSID_SW_TYPE Value:10.0
 Key:SYSID_THISMAV Value:1.0
 Key:SYSID_MYGCS Value:255.0
 Key:CLI_ENABLED Value:0.0
 Key:PILOT_THR_FILT Value:0.0
 Key:PILOT_TKOFF_ALT Value:0.0
 Key:PILOT_TKOFF_DZ Value:100.0

...
```

Could you please post the output of `pip list`? I suspect that the problem could be an outdated version of pymavlink. Also, are you using the latest master of DroneKit? Here's mine:

```
$ pip list
Package       Version Location
------------- ------- --------------------------------------
dronekit      2.9.1   /Users/pietro/FlyToons/dronekit-python
dronekit-sitl 3.2.0
future        0.17.1
lxml          4.3.1
mock          2.0.0
monotonic     1.5
nose          1.3.7
pbr           5.1.2
pip           18.1
psutil        5.5.1
pymavlink     2.3.4
setuptools    40.6.2
six           1.12.0
```

pietrodn, 2019-02-24:
I think your bug has been fixed by [this commit](https://github.com/ArduPilot/pymavlink/commit/4026f1df0f2bfd0925e307a34af7eaa348e0597d#diff-ca22e0f769f9965994135668c57d3f79R774) in pymavlink, which decodes the bytestring parameter names into strings under Python 3.

custeve, 2019-02-25:
Pietro, thanks for looking into this. Yes, I was running pymavlink 2.0.6. This is the version installed in setup.py for dronekit. When you upgrade, it reports: 
```
dronekit 2.9.1 has requirement pymavlink==2.0.6, but you'll have pymavlink 2.3.4 which is incompatible.
````
I upgraded to pymavlink 2.3.4 and everything appears to be working. Printing parameters, setting a parameter and the parameter listener worked as it should. Thanks! 

So - At this point, the issue seems to be the dependence on 2.0.6 in the pip installer. I was having a lot of issues getting dronekit working for python 3 initially, and I ran several different installs at the time, including both pip and cloning. Looks like I must have last installed with pip and thought it was fine.

pietrodn, 2019-02-25:
The issue here is quite complex: the current master has the correct version requirement for `pymavlink`, but it depends on `dronekit_sitl` which depends on PyPI's `dronekit` which depends on the old pymavlink... this mess should clear once a new stable version of `dronekit` is released on PyPI.

Happy to see it's working for you, though! :D

pietrodn, 2019-03-05:
@custeve can this issue be closed?

custeve, 2019-03-05:
@pietrodn, that is a good question. With the correct pymavlink, things have been working great. Is there an existing issue that calls to resolve the dependency in setup.py for python3 installations? If not, I'd say either this should stay open or we create a new issue for it.

pietrodn, 2019-03-05:
The current master of DroneKit already has as a requirement in `setup.py`: `pymavlink>=2.2.20`.

Issue #909 asking for a new stable release is already open.