dronekit/dronekit-python #909: Cut new release?

drnic, 2019-02-13:
https://github.com/dronekit/dronekit-python/releases/tag/v2.9.1 was cut April 2017

What are the TODO items for a v2.9.2 or v2.10 or v3 release? (which would include the python3 support #891)

hamishwillee, 2019-02-13:
@peterbarker is probably the best person to answer that!

peterbarker, 2019-03-14:
I was thinking we might just from the current master - pretty much as soon as either (a) there's a chorus of "Yes please!") or (b) there's a deafening silence or.

https://github.com/dronekit/dronekit-python/pull/919 <- that would be merged into master and the v2.9.2 tag created.

Do we need #907 in first?

peterbarker, 2019-03-14:
https://test.pypi.org/project/dronekit/#files <- test packages for 2.9.2

... and while we're at it: https://test.pypi.org/project/dronekit-sitl/#files