dronekit/dronekit-python #912: Issues running the examples on the Intel Aero RTF Drone

AauSurfaceGo, 2019-02-20:
Hello,
I am currently trying to perform autonomous flying development for my  but I am having some problems running the examples from this repository on my. Every time i try to run any of the examples i get either timeout due to lack of connection after the 

>>"Connecting to vehicle on: tcp:127.0.0.1:5760" 

appears on my terminal, or i get 
>>data link #0 lost
>>data link #0 regained 

and then the drone fails to run the code file. 

I have been able to do an autonomous take off using QGroundControl but I am unable to run any python script on the drone. 
I hope someone can shed some light on the subject.
Thank you in advance,

peterbarker, 2019-03-13:
On Wed, 20 Feb 2019, AauSurfaceGo wrote:

> I am currently trying to perform autonomous flying development for my but I am having some problems running the examples from this repository on my. Every
> time i try to run any of the examples i get [...]

It's trying to connect to a simulated aircraft.

You need to supply a connect string - try "--help" on the examples - but 
you'll probably just specify the path to the serial port...

ivanalexo, 2019-04-25:
Which firmware of Flight controller did you install in your Intel RTF drone? Px4 or ardupilot?