dronekit/dronekit-python #913: Cannot connect while using MAVROS

jwen18, 2019-02-22:
We are running MAVROS to get the external position data from a SLAM algorithm. We are using APM3.5.5. However, once I startup MAVROS, my Dronekit-Python scrip which controls a drone in GUIDED_NO_GPS mode can no longer connect. I get an error that says "mode 0 not available on mavlink definition". Does Dronekit not work while MAVROS is being used?

amilcarlucas, 2019-02-23:
dronekit will only work if you connect it to the GCS udp port of mavros.
That way both can work in parallel.
This is not a bug, and should be closed.

shubham-shahh, 2021-11-18:
> dronekit will only work if you connect it to the GCS udp port of mavros. That way both can work in parallel. This is not a bug, and should be closed.

Hi, @amilcarlucas how can we connect dronekit to the GCS udp port of mavros

amilcarlucas, 2021-11-22:
Read the mavros documentation on the GCS output. It is basically a UDP connection to port 14550