dronekit/dronekit-python #915: when i run fake gps i got error

nagendra092918, 2019-03-05:
rohan@rohan-HP-15-Notebook-PC:~$ dronekit-python/examples/follow_me/run-fake-gps.shSending fake GPS data
Traceback (most recent call last):
  File "/usr/bin/gpsfake", line 25, in <module>
    import gps.fake as gpsfake   # The "as" pacifies pychecker
  File "/home/rohan/.local/lib/python2.7/site-packages/gps/fake.py", line 90, in <module>
    from . import packet as sniffer
ImportError: cannot import name packet

please help.thank you

peterbarker, 2019-03-13:
On Tue, 5 Mar 2019, NAGENDRA BABU MARELLA wrote:

> File "/home/rohan/.local/lib/python2.7/site-packages/gps/fake.py", line 90, in
> from . import packet as sniffer
> ImportError: cannot import name packet

Not really a dronekit-python qeustion :-)

You're missing something like the Python packet python-socketio

Peter

nagendra092918, 2019-03-22:
@peterbarker how to find that

nagendra092918, 2019-03-22:
@peterbarker  is dronekit-python will work on python3 please help if yes what are change i should do please help

jmachuca77, 2019-03-22:
This is a support issue please use the forums to post support questions. Thanks

closed 2019-03-22 by jmachuca77