dronekit/dronekit-python #921: how to use Dronekit python with python3 when i am trying run with python 3 i got a error

nagendra092918, 2019-03-21:
[image]
[image]

AdamHtooLwin, 2019-04-17:
Did you install to python or python3? Besides, dronekit isn't supported on Python 3 I believe.

nagendra092918, 2019-04-18:
i installed python and python3.here python2 working fine on dronekit but python3 not working
but in pip install they given supported for python3 please check once image also if have any idea please tell to me . i tried alot
[image]

closed 2019-04-30 by nagendra092918

wrmichaelmak, 2019-08-06:
Are you able to solve this problem?

leonardocfor, 2019-08-06:
Nagendra, why are you using sudo to run the simple_goto script? Try running it without sudo. It should work. Otherwise can you show the source code of the script please?