dronekit/dronekit-python #922: Cannot connect to drone between raspberry pi 3 B and pixhawk

ekaratst, 2019-03-29:
I cannot connect between raspberry pi model 3 B and pixhawk. I'm trying to run this code and i also enable_uart.

`from dronekit import connect
vehicle = connect('/dev/ttyS0', baud=57600, wait_ready=True)`

This is error.
`ERROR:dronekit.mavlink:Exception while receiving message:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 218, in mavlink_thread_in
    msg = self.master.recv_msg()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 411, in recv_msg
    s = self.recv(n)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 955, in recv
    ret = self.port.read(n)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 490, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
`
How should I do?

borhanreo, 2019-04-04:
Please check your tty port 
`ls -l /dev/tty*
`
I think it will be
`vehicle = connect('/dev/ttyACM0', wait_ready=False)`

ekaratst, 2019-04-09:
> @ekaratst Please check your tty port
> `ls -l /dev/tty* `
> I think it will be. you can try this
> `vehicle = connect('/dev/ttyACM0', wait_ready=False)`

I check port and get result below.
"""
crw-rw-rw- 1 root tty       5,  0 Apr  9 12:17 /dev/tty
crw--w---- 1 root tty       4,  0 Apr  9 12:17 /dev/tty0
crw--w---- 1 root tty       4,  1 Apr  9 12:29 /dev/tty1
crw--w---- 1 root tty       4, 10 Apr  9 12:17 /dev/tty10
crw--w---- 1 root tty       4, 11 Apr  9 12:17 /dev/tty11
crw--w---- 1 root tty       4, 12 Apr  9 12:17 /dev/tty12
crw--w---- 1 root tty       4, 13 Apr  9 12:17 /dev/tty13
crw--w---- 1 root tty       4, 14 Apr  9 12:17 /dev/tty14
crw--w---- 1 root tty       4, 15 Apr  9 12:17 /dev/tty15
crw--w---- 1 root tty       4, 16 Apr  9 12:17 /dev/tty16
crw--w---- 1 root tty       4, 17 Apr  9 12:17 /dev/tty17
crw--w---- 1 root tty       4, 18 Apr  9 12:17 /dev/tty18
crw--w---- 1 root tty       4, 19 Apr  9 12:17 /dev/tty19
crw--w---- 1 root tty       4,  2 Apr  9 12:17 /dev/tty2
crw--w---- 1 root tty       4, 20 Apr  9 12:17 /dev/tty20
crw--w---- 1 root tty       4, 21 Apr  9 12:17 /dev/tty21
crw--w---- 1 root tty       4, 22 Apr  9 12:17 /dev/tty22
crw--w---- 1 root tty       4, 23 Apr  9 12:17 /dev/tty23
crw--w---- 1 root tty       4, 24 Apr  9 12:17 /dev/tty24
crw--w---- 1 root tty       4, 25 Apr  9 12:17 /dev/tty25
crw--w---- 1 root tty       4, 26 Apr  9 12:17 /dev/tty26
crw--w---- 1 root tty       4, 27 Apr  9 12:17 /dev/tty27
crw--w---- 1 root tty       4, 28 Apr  9 12:17 /dev/tty28
crw--w---- 1 root tty       4, 29 Apr  9 12:17 /dev/tty29
crw--w---- 1 root tty       4,  3 Apr  9 12:17 /dev/tty3
crw--w---- 1 root tty       4, 30 Apr  9 12:17 /dev/tty30
crw--w---- 1 root tty       4, 31 Apr  9 12:17 /dev/tty31
crw--w---- 1 root tty       4, 32 Apr  9 12:17 /dev/tty32
crw--w---- 1 root tty       4, 33 Apr  9 12:17 /dev/tty33
crw--w---- 1 root tty       4, 34 Apr  9 12:17 /dev/tty34
crw--w---- 1 root tty       4, 35 Apr  9 12:17 /dev/tty35
crw--w---- 1 root tty       4, 36 Apr  9 12:17 /dev/tty36
crw--w---- 1 root tty       4, 37 Apr  9 12:17 /dev/tty37
crw--w---- 1 root tty       4, 38 Apr  9 12:17 /dev/tty38
crw--w---- 1 root tty       4, 39 Apr  9 12:17 /dev/tty39
crw--w---- 1 root tty       4,  4 Apr  9 12:17 /dev/tty4
crw--w---- 1 root tty       4, 40 Apr  9 12:17 /dev/tty40
crw--w---- 1 root tty       4, 41 Apr  9 12:17 /dev/tty41
crw--w---- 1 root tty       4, 42 Apr  9 12:17 /dev/tty42
crw--w---- 1 root tty       4, 43 Apr  9 12:17 /dev/tty43
crw--w---- 1 root tty       4, 44 Apr  9 12:17 /dev/tty44
crw--w---- 1 root tty       4, 45 Apr  9 12:17 /dev/tty45
crw--w---- 1 root tty       4, 46 Apr  9 12:17 /dev/tty46
crw--w---- 1 root tty       4, 47 Apr  9 12:17 /dev/tty47
crw--w---- 1 root tty       4, 48 Apr  9 12:17 /dev/tty48
crw--w---- 1 root tty       4, 49 Apr  9 12:17 /dev/tty49
crw--w---- 1 root tty       4,  5 Apr  9 12:17 /dev/tty5
crw--w---- 1 root tty       4, 50 Apr  9 12:17 /dev/tty50
crw--w---- 1 root tty       4, 51 Apr  9 12:17 /dev/tty51
crw--w---- 1 root tty       4, 52 Apr  9 12:17 /dev/tty52
crw--w---- 1 root tty       4, 53 Apr  9 12:17 /dev/tty53
crw--w---- 1 root tty       4, 54 Apr  9 12:17 /dev/tty54
crw--w---- 1 root tty       4, 55 Apr  9 12:17 /dev/tty55
crw--w---- 1 root tty       4, 56 Apr  9 12:17 /dev/tty56
crw--w---- 1 root tty       4, 57 Apr  9 12:17 /dev/tty57
crw--w---- 1 root tty       4, 58 Apr  9 12:17 /dev/tty58
crw--w---- 1 root tty       4, 59 Apr  9 12:17 /dev/tty59
crw--w---- 1 root tty       4,  6 Apr  9 12:17 /dev/tty6
crw--w---- 1 root tty       4, 60 Apr  9 12:17 /dev/tty60
crw--w---- 1 root tty       4, 61 Apr  9 12:17 /dev/tty61
crw--w---- 1 root tty       4, 62 Apr  9 12:17 /dev/tty62
crw--w---- 1 root tty       4, 63 Apr  9 12:17 /dev/tty63
crw--w---- 1 root tty       4,  7 Apr  9 12:29 /dev/tty7
crw--w---- 1 root tty       4,  8 Apr  9 12:17 /dev/tty8
crw--w---- 1 root tty       4,  9 Apr  9 12:17 /dev/tty9
crw-rw---- 1 root dialout 204, 64 Apr  9 12:29 /dev/ttyAMA0
crw------- 1 root root      5,  3 Apr  9 12:17 /dev/ttyprintk
crw--w---- 1 root tty       4, 64 Apr  9 12:31 /dev/ttyS0
"""
Also,  I change wait_ready=False. A result it's same.

nagendra092918, 2019-04-18:
check the serial port and import serial port u may get positive result

ekaratst, 2019-04-21:
Thank you everybody. I found a solution. Change serial port to micro usb and use /dev/ttyACM0.

closed 2019-04-21 by ekaratst